robotology/mesh-support #884: Some part of the skin are broken

S-Dafarra, 2019-11-13:
## Brief description of the request/failure
Some skin parts do not seem to be working. In particular:
- some portions of the torso are not sensitive
- the right arm boards do not seem to be running
- the back part of the left forearm does not seem to be working

This is in preparation for the RAI demo, hence I  would address the easy fixes. If the fix would require to unmount several parts of the robot and/or keep it busy for a long time, I would postpone it after the demo.

## Detailed context

## Accompanying material

cc. @fjandrad

davidetome, 2019-11-16:
2 boards (mc4plus and mtb3) have been replaced; see tickets below:

https://github.com/icub-tech-iit/tickets/issues/153
https://github.com/icub-tech-iit/tickets/issues/154

cc @julijenv  @maggia80  @Fabrizio69

S-Dafarra, 2019-11-19:
The right hand appeared to have a lot of problems. The problem was that MAIS board was not communicating. @julijenv noticed that several boards where shortcircuited and had to remove almost all the skin boards. Also, he noticed that the pinky spring was broken. All these problems were temporarily fixed to run the demo.

@julijenv do you want me to open another issue to definitely fix the problems above?

julijenv, 2019-11-27:
it has been checked and nothing can be

closed 2019-11-27 by julijenv