robotology/mesh-support #885: Installation of the icub live

TheCamusean, 2019-11-13:
## Brief description of the request/failure

Hello, I am trying to reinstall the Operative system in the icub head and I am dealing with some problems. What I did is : 

1. Download the latest icub live version : icub-live 7.10 from http://www.icub.org/download/pc104-images/

2. I format my usb stick in FAT and copy the data inside the zip folder in the root of the usb stick.

3. I went to the /mounting_point/utils/linux and sudo su
4. I called sudo bash makeboot.sh /dev/sdb1
5. Say yes to everything in the command line
6. plug the usb stick on the icub's head

## Detailed context

Once I plug the system runs but it cannot boot. 

The error  is 
[image]

Any suggestions??

## Accompanying material

julijenv, 2019-11-13:
Hi @TheCamusean ,
some questions just popped into my head that are much than just useless.
which robot are we talking about? CAN - ETH protocol?
what kind of USB stick are you using? USB2 or USB3 -> the icub-head or pc104 DONT support USB3
did you read this page
http://wiki.icub.org/wiki/The_Linux_on_the_pc104
let me know

dcam0050, 2019-12-10:
@TheCamusean Can you provide details as to how you solved this issue as we have encountered the exact problem with iCubSheffield01. We are updating iCub's live image on the same usb that contained the previous image so that rules out issues with USB and USB3 incompatibility.

dcam0050, 2019-12-10:
For anyone encountering this issue, it seems to be a problem with ldlinux.c32. This issue was fixed by downloading the live cd iso for debian buster and copying /debian-iso/isolinux/ldlinux.c32 to /icubUsb/syslinux

closed 2019-12-10 by dcam0050