robotology/mesh-support #888: Right ring finger misplaced

xEnVrE, 2019-11-19:
## Brief description of the failure

When the `ring and little finger` joint of the right hand is commanded to its lower bound, the ring finger assumes a wrong position as depicted in the picture. If the desired joint is moved slightly away from the lower bound, the ring finger recovers its position.

[image]

eleramp, 2019-11-29:
Hi @julijenv, any news about this? Sorry for bothering you, we would need this to be fixed asap, in order to be able to do the demos on December 9th.

julijenv, 2019-11-29:
@davidetome , puoi mica andare a verificare che magari ci sia soltanto da ricalibrare?!? Grazie in anticipo

davidetome, 2019-11-29:
@julijenv  just checked;  the cable was wrappred in the bad way.
I wrapped it back in the right direction using the screwdriver.

please, @eleramp can you check the encoder reading of the joint in the motorgui when you come back?

eleramp, 2019-11-29:
Thank you @davidetome, it works well! Encoder reading are okay.

eleramp, 2019-12-03:
I'm sorry I was wrong. After few days of usage the finger is again as in the picture.

julijenv, 2019-12-03:
It should be ok, now, I mean at least the motor will not spin back and close the rig finger. It would be nice to understand if the CAN board MC4 have a FW compatible with controltype 12, Could you @valegagge confirm that to me, because as iCubGenova01 is the only robot with no xls files associated ( it has been deleted at some point by mistake and never retrieved).
let us know.

eleramp, 2019-12-05:
Hi, I try to use the right hand today, for the first time after Julien's fix. Unfortunately, the ring and little finger joint does not move, neither by controlling its position from the motorgui..

fbottarel, 2019-12-09:
Any news on this? @julijenv repaired the fingers but the resting position of the joints show up as values that are different from zero in the motorgui. Shall we fix the configuration issue and missing files problem once and for all? @valegagge 

@lornat75

pattacini, 2019-12-10:
With a visual inspection, we can confirm that fingers are working properly.

Thanks @emilianob80 and @fiorisi

closed 2019-12-10 by pattacini