robotology/mesh-support #889: iCubV3 short circuit on PC104 power supply input

ale-git, 2019-11-20:
## Brief description of the request/failure

The power supply board goes in fault state as soon as the PC104 is switched on.

## Detailed context

When the PC104 board is unplugged there is no fault on CPU power supply. Moreover, the PC104 impedence between the terminals of the PC104 power supply connector is less than 1 ohm.  

@maggia80 @pattacini

julijenv, 2019-11-20:
Double check, there is a shortcircuit in between the input pins of the pc104.
The board need to be replaced I presume, a further and deeper check need to be done on the connection inside the conneections side of the pc104 to see if it is not one of the peripheral that put the pc104 in short.
more coming

maggia80, 2019-11-20:
@julijenv yes I would suggest unplugging all the connection between the COM-Express Carrier board ([CCG012](http://connecttech.com/product/com-express-type-6-rugged-ultra-lite-carrier-board/) and all the peripherals. 

[image]

closed 2019-11-21 by julijenv