robotology/mesh-support #890: Purple iCub idling wrists when turning on

marco-monforte, 2019-11-21:
## Brief description of the request/failure
Hey guys,
after turning on the yarprobotinterface and the calibration procedure, both left and right wrist are disabled and fall down.
As you know, on December 9th we have the visit of the IIT Scientifical and Technical Committee. It would be nice to fix this small issue. Thanks!

## Accompanying material

[image]

marco-monforte, 2020-06-05:
Ciao! Next week (no precise day) Giulia needs the robot to start some experiments. Could we fix the wrists when you have some time?

violadelbono, 2020-06-09:
Hello @marco-monforte 

The problem was due to the parameter `startupPosThreshold` in the calibration files of the arms. The threshold of joints 5 and 6 was initially set too low and thus the robot could not hold its wrists in position after the initial calibration. 

We changed the threshold from 2 to 90, according to the configuration files of the previous robots.

marco-monforte, 2020-06-09:
Thanks @violadelbono and @julijenv !

I think for the time being is the best we can have. In the future, we could investigate why the wrists cannot reach an accurate delta threshold during the calibration stage (probably because of some PWM issue?).

closed 2020-07-20 by julijenv