robotology/mesh-support #891: Left wrist and pronosupination broken

S-Dafarra, 2019-11-21:
## Brief description of the request/failure
The left wrist is not calibrating and the logger throws many errors about overcurrent and i2c errors.

## Detailed context
We were preparing for a demo when the left foot stopped calibrating even after several reboots. It seems to be an electrical problem or a motor burnt.

## Accompanying material

S-Dafarra, 2019-11-21:
I took several hours to @julijenv to figure out that probably a connector of the finger adduction joint was somehow flaky and it was giving problems. By fixing it, it started working again. 

The irony is that the finger adduction is pretty useless for our demos, yet it prevented us from showing them. The issue was that neither the wrist pitch and yaw were calibrating. By taking a look at the configuration files, it turned out that this joint (the number 7) is calibrating together with the two joints of the wrist (joint 5 and 6): https://github.com/robotology/robots-configuration/blob/devel_iCubGenova04/iCubGenova04/calibrators/left_arm-calib.xml#L36

Hence, an overcurrent problem in this joint was preventing the other two to start. Probably by separating joints 5 and 6 from the 7th during calibration can avoid this problem to be blocking in the future.

Nevertheless, joint 4 (pronosupination) was also problematic, having issues during calibration (it was moving slower during calibration) or in a weird way. @julijenv do you think it may be related to the faulty connector?

DanielePucci, 2019-11-21:
CC
@GiulioRomualdi @MiladShafiee @kouroshD @Yeshasvitvs @lrapetti @nunoguedelha @claudia-lat @prashanthr05

S-Dafarra, 2020-01-21:
This has been fixed. Thanks @julijenv

closed 2020-01-21 by S-Dafarra