robotology/mesh-support #893: iCubDarmstadt01 : yarprobotinterface not working

TheCamusean, 2019-11-23:
## yarprobotinterface not working after a reinstallation of the OS in the robot

Hello, I have installed Yarp v3.2.1 and Icub v1.13.0 on our robot in Darmstadt.

After the installation, yarprobotinterface is not working.

I attach some of the error messages in the log

[image]
[image]
[image]

I hope you can help me :)

julijenv, 2019-11-25:
Dear @TheCamusean , could you please send a full logfile please, thx in advance

julijenv, 2019-11-27:
ping @TheCamusean

julijenv, 2019-12-03:
Desar @TheCamusean ,
You need to update the FW of the EMS board inside the robot from what I can see, I would have loved to see the entire log but looks like you may alreasdy figure out what's going on.
You need to use FirmwarUpdater.
You probably need to contact me to help you to do so.
let me know - my skype-I is icub.support

TheCamusean, 2019-12-03:
Hi @julijenv , 

The following weeks I will be really busy, but maybe we can contact some time in the future and finish this work.

Thanks alot for the support

closed 2020-02-08 by stale[bot]