robotology/mesh-support #895: FT sensor on a test-bench not readable after STRAIN2 firmware update

HosameldinMohamed, 2019-11-29:
## Brief description of the request/failure
I have tried to update the firmware of an FT sensor (mounted on the turbines test-bench) and now I am not able to detect the board using `FirmwareUpdater`.
STRAIN2 board before update: `1.3.8`
The version intended to update: `2.0.4`
Device used: USB2CAN converter.

## Detailed context
- I have downloaded this file (https://github.com/robotology/icub-firmware-build/blob/3c96c54fa435e9f5cdb0b1e434129d99a57930f8/CAN/strain2/strain2.hex). and opened `FirmwareUpdater`.
- I selected the board and clicked on "Upload Application" and selected the mentioned file.
- The uploading took some time and finished with no noted issues. I saw that the version didn't change from the previous value so I restarted the `FirmwareUpdater` and unplugged/plugged the device.
- After, I am not able to detect the device anymore.

- I get the following message on the terminal when pressing "Discover" button:
```
[INFO]created device <socketcan>. See C++ class yarp::dev::SocketCan for documentation.
Error: SocketCan::canWrite() was unable to send message.
Error: SocketCan::canWrite() not all messages were sent.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
[INFO]created device <socketcan>. See C++ class yarp::dev::SocketCan for documentation.
Error: SocketCan::canWrite() was unable to send message.
Error: SocketCan::canWrite() not all messages were sent.
FirmwareUpdaterCore::getCanBoardsFromDriver(): No answer received from CAN boards after a successful driver init.
```

**Note**: I usually run these bash lines after plugging the device:
```
echo executing ESDcan commands
sudo ip link set can0 type can bitrate 1000000
sudo ip link set up can0

echo Verify correct values for FT/sense 
echo address on jet setup is 0x02
echo Device number is Net0
```

## Accompanying material

Here's the `FirmwareUpdater` GUI during the uploading:
[image]

HosameldinMohamed, 2019-11-29:
cc @marcoaccame

traversaro, 2019-11-29:
Are you sure that the ft sensor is equipped with the STRAIN2 board, and not the older STRAIN board?

HosameldinMohamed, 2019-11-29:
> Are you sure that the ft sensor is equipped with the STRAIN2 board, and not the older STRAIN board?

Yes, judging from the screenshot in the description, because this is the information also before the upload.

marcoaccame, 2019-11-29:
Yes, it is a strain2. The gui clearly shows it.

@HosameldinMohamed, you have been very careful to save the print screen. Very well done.

Unfortunately, this strain2 still has an early prototype firmware version 1.x.x.

I was not aware that there were other strain2 boards with version  1.x.x still around. I personally did around about March the migration to version 2.x.x of what I thought to be every strain2 around: those on genova04/02 and on some shoes.

See https://github.com/robotology/icub-firmware-build/blob/master/CAN/strain2/128kmapping/howto.migrate.from128kmapping.txt for how to do it.

Now, however, the recent upgrade attempt has written onto the bootloader space and this procedure cannot be applied anymore.

the best way to proceed is to reprogram bootloader of the board with jtag. Then with firmwareupdater to load the application and the regulation set of SN 163.

The guys from task force will be able to take care of that on Monday, and in case of need I will be keen to supporters them.

HosameldinMohamed, 2019-12-03:
@marcoaccame should I ask someone from the task force personally for help? Thanks.

pattacini, 2019-12-05:
cc @julijenv @triccyx

HosameldinMohamed, 2019-12-06:
Thank you @triccyx and @marcoaccame. Now the sensor is alive again and up-to-date :+1: 

[image]

closed 2019-12-06 by pattacini