robotology/mesh-support #896: Skin hand taxel mapping

lucagl, 2019-12-03:
Hi all,
With @dabbous, @arrenglover, and @ChiaraBartolozzi I am trying to build a visualizer for the event-driven library with a nice map of the skin events. In particular, we would like to have it working on events coming from the hand skin of the iCub. 
I found in the icub-main library methods and files to print and read the skin map. Combining .ini and .txt (calibration) files, I'm able to find how actual taxel addresses are mapped for some skin patches (eg. triangle 10 pad). To do so I used some a priori knowledge on the actual location of each taxel (see figure below) which helped me to device how the numbering is built. The files themselves are indeed extremely hermetic to me (those in icub-main/app/skinGui/conf). 
With the hand patch I'm halfway in the job. I figured out how to map the fingers but still the palm is puzzling me. Do you have a map of the type of the figure below relating taxel addresses and position for the hand? This would help a lot.

Thank you

Luca

[image]

julijenv, 2019-12-03:
@maggia80 to whom can I ask for feedback about that?!?! thx in advance

simeonedussoni, 2019-12-03:
@julijenv tomorrow I will provide the documentation about this.

simeonedussoni, 2019-12-05:
[image]
here the mapping of the taxels in the fingertips, left and right respectively.
The correspondence between taxels and data is as follows
- data are streamed in MTB(3/4) as two packets of 8 bytes
- first packet contains header 40 + 7 words of data from taxels 0-6
- second packet contains header C0 + 5 words o data from taxels 7-11  + 2 control bytes
- this structure is replicated for each AD7147 chip in the I2C bus ( addresses from 0 to 3) and for each I2C bus from SDA1 to SDA3 

the taxel positions for the palm are represented in the following figure
[image]
please note that you need to add the correct offset for finding the taxels in the data stream.

lucagl, 2019-12-05:
@simeonedussoni  any chance for the documentation coming soon? 
thank you
Luca

lucagl, 2019-12-05:
thank you very much!

julijenv, 2019-12-20:
thx to @simeonedussoni

closed 2019-12-20 by julijenv