robotology/mesh-support #897: i2c-3 line malfunction for ZCB to ED-MTB connection

arrenglover, 2019-12-05:
## Brief description of the request/failure

The i2c-3 line on the ZCB (event-driven board) is scrambled when the yarprobotinterface is running.

## Detailed context

The i2c-3 line on the ZCB is used to connect to the ED-MTB board for event-driven skin. Currently it is connected via a cable that runs down the left arm of the robot to the left hand of the robot. When the yarprobotinterface is off we can successfully connect to the ED-MTB board via i2c protocol. A stable address 1e is correctly shown when running
```
sudo i2cdetect -y 3
```
on the icub-zynq board.

When the yarprobotinterface is turn on (and the robot is actively controlled), instead we cannot connect to the ED-MTB board and the 1e address is no longer stable

This issue has occurred before, I think the solution was to correctly insulate the cable such that motor noise would not affect the signals on the i2c line. Perhaps the cable has worn, moved, or damaged. 

@dabbous94 can be contacted to help with finding the issue (i.e. running the `i2cdetect` command correctly)

@ChiaraBartolozzi may have other information about how we solved the problem before.

emilianob80, 2019-12-06:
@dabbous94 please can you contact me so let's see the problem together. Thanks

cc: @ChiaraBartolozzi @arrenglover

emilianob80, 2019-12-06:
The wiring diagram used for electrical wiring is as follows:

[ZCB wiring.pdf](https://github.com/robotology/icub-tech-support/files/3930793/ZCB.wiring.pdf)

The shield connection was not foreseen and therefore, if the problem is caused by motor disturbances, I believe it is necessary to provide a shield connected to the GND pin on the head side. @MrAndrea  what do you think?

I apologize for my English

emilianob80, 2019-12-06:
A verification on the robot has already highlighted the shield connected to GND even if not requested as a diagram. At this point I will try to re-crimp all the wires and reduce the length of the cable since it is present on the torso of abundance

I apologize for my English

arrenglover, 2019-12-06:
Grazie @emilianob80!

I don't know why the problem started again, and couldn't identify the cause myself. Thanks for looking into it.

emilianob80, 2019-12-06:
@arrenglover I crimped all the wires again and tested the continuity. Everything seems OK. When you arrive if you call me we will do some tests together.
Thanks

arrenglover, 2019-12-06:
@dabbous94 - are you here today? Can you check the skin if the skin is now working correctly?

dabbous94, 2019-12-06:
> @dabbous94 - are you here today? Can you check the skin if the skin is now working correctly?

yes, @arrenglover, Emiliano is here and we are working on solving the issue.

emilianob80, 2019-12-06:
@arrenglover The skin works correctly at first boot but, in case of reboot with engines running, reboot fails. We did some tests and came to the conclusion that the left shoulder is the cause of the problems. With the left shoulder disabled the skin starts regularly. This problem cannot be solved at this time due to time constraints. I can, for the sake of scrupulousness, put a ferrite on my hand to see if we are lucky and the noise is reduced. For the demo, in case of skin restart, do it with the left shoulder disabled.
cc: @ChiaraBartolozzi

arrenglover, 2019-12-06:
Thanks very much. That sounds good. We can work around the problem for the demo next week now you found the exact problem.

@emilianob80 - can you please put the icub head cover on the robot as soon as you can, I have to make a video of the robot this afternoon.

emilianob80, 2019-12-06:
> Thanks very much. That sounds good. We can work around the problem for the demo next week now you found the exact problem.
> 
> @emilianob80 - can you please put the icub head cover on the robot as soon as you can, I have to make a video of the robot this afternoon.

@arrenglover 
- head cover ok
- ferrite has no benefits

arrenglover, 2020-02-04:
@emilianob80 @dabbous94 @ChiaraBartolozzi - we should probably come up with a permanent solution to this problem

closed 2020-08-16 by stale[bot]