robotology/mesh-support #899: Align right shoulder's yaw with the left counterpart

pattacini, 2019-12-07:
## Brief description of the request/failure
While setting up iCub 3, we realized that the right shoulder yaw is mounted with a physical displacement of roughly 20 deg with respect to the left yaw.

As of now, the software accounts for the offset, but it'd be good to fix this up.

julijenv, 2019-12-08:
I think it would be done later, when the visit with the Committee will be ended, I dont think you want to do that before, right? @pattacini

pattacini, 2019-12-08:
Yep, there's no hurry 👍

pattacini, 2020-01-21:
After the maintenance, we need to remember to change the joints' bounds in the configuration file.

@julijenv

julijenv, 2020-01-21:
@marcellosavoldi

marcellosavoldi, 2020-01-21:
I'm checking the CAD to understand where is the problem.
For now, is all right and the right arm met to the CAD
I need to investigate better on the robot.

[image]

marcellosavoldi, 2020-01-21:
Tomorrow I'll remove the left arm...
I have to check the right position of the plug for hard-stop.
Is possible that the wrong position of the arm is on the left arm and no on the right.

[image]

marcellosavoldi, 2020-01-22:
@pattacini  and @julijenv ,
I removed the left arm and I noted that there is a problem with the supports fixed on the FTSensor on the arm:

[image]

The part and the drawing are different:

[image]

I'm sorry, but I don't remeber the history of the part.
I think that we need to open a workshop job to create the right part:
Q.ty 2 code ICUB3_001_P_234 (7082.A2)

Fabrizio69, 2020-01-22:
@pattacini ho richiesto una commessa d'officina per la produzione dei due pezzi richiesti da @marcellosavoldi (vedi *[Proto issue #220](https://github.com/icub-tech-iit/proto/issues/220)*.
Vi segnalerò quando arriveranno i pezzi a magazzino

pattacini, 2020-01-23:
While waiting for the parts to be produced, we will mount the arms in the current (buggy) configuration.

Fabrizio69, 2020-02-05:
@pattacini @julijenv @marcellosavoldi il materiale richiesto è disponibile presso il Proto Team Warehouse (vedi *[Proto issue #243](https://github.com/icub-tech-iit/proto/issues/243)*)

pattacini, 2020-02-05:
Thanks @Fabrizio69 👍 
@julijenv will be dealing with all the necessary steps, as agreed f2f.

julijenv, 2020-02-07:
thx @Mick3Lozzo from task-force to have handled the repair.
we still need @ale-git to handle the recalibration and mainly let us know if the repair went fine.

pattacini, 2020-02-08:
@ale-git did it in https://github.com/robotology/robots-configuration/pull/166 👍

closed 2020-02-08 by pattacini