robotology/mesh-support #901: Unable to communicate with the head using the wifi

GiulioRomualdi, 2019-12-10:
## Brief description of the request/failure
The connection between the iCubGenova04 and the router wifi seems to be not stable. Indeed yesterday we (@kouroshD) tried to connect the robot to the local thorough the wifi. After a couple of minutes we were not able to reach the head (neither `ssh icub-head` nor `ping icub-head`). 
Connecting the ethernet cable fixed the problem and it was possible to reach the head. 

## Accompanying material
Unfortunately none

julijenv, 2019-12-18:
under test @GiulioRomualdi , give me feedbacks

julijenv, 2019-12-20:
didnt find any problem there when passing by, leave it on hold for the moment.

S-Dafarra, 2020-01-16:
For the time being, we can close this issue. We can reopen it if it happens again, but unfortunately, it is very random.

closed 2020-01-16 by S-Dafarra