robotology/mesh-support #904: Test Microphones: four seconds delay

linzdg, 2020-01-01:
When I followed the tutorial with the test microphones,

1. On the server type in a terminal
    icub@icubsrv:~$ yarpdev --device portaudioPlayer
2. On the head type in a terminal
    icub@icubsrv:~$ yarpdev --device portaudioRecorder --start
3. On the server type in a terminal
    icub@icubsrv:~$ yarp connect /audioRecorderWrapper/audio:o /audioPlayerWrapper/audio:i

There was a delay of almost four seconds from the sound caught by the microphones on the head’s ears to hearing on the server speakers. How to solve this problem?

pattacini, 2020-01-02:
@randaz81 raised the following warning in #774:

>Pease remember that a continuous, realtime audio streaming (like skype) is not possible using YARP. It implements only basic buffering and UDP communication. So received audio will be subject to a constant delay, determined by the size of the buffer.

Thereby, @randaz81, could you please help us understand here whether the delay @linzdg is experiencing is somehow expected - although quite huge - or there might be other factors affecting iCubShenzhen01?

randaz81, 2020-01-14:
As mentioned by @pattacini, YARP currently does not support **RTP (Real Time Protocol)**. For this reason, streaming audio via TCP/UDP requires **buffering**. By default a buffer of approx 5 seconds is set by **AudioPlayerWrapper**, which is responsible of receiving data from **AudioRecorderWrapper**. When the buffer is full, the wrapper will start the playback (similary to YouTube buffering technique).

You can change the size of the buffer with the option `--playback_network_buffer_lenght`. 
For example: `yarpdev --device portaudioPlayer --playback_network_buffer_lenght 1` will set the size of the buffer to 1 second. 

However, we do not recommend reducing the buffer size, since network transmission has unpredictable jitter. Since delays accumulate over time, buffer underruns are inevitable. This will generate audible noise (clicks) during the playback and will corrupt the audio waveform.

Nicogene, 2020-01-16:
Hi @linzdg,

Coluld you paste here your version of yarp?

pattacini, 2020-01-22:
Closing, feel free to open it up again if needed.

closed 2020-01-22 by pattacini

farooqUsman, 2022-10-11:
@randaz81 Hello
I wanted to access the ear of ICUB so kindly
guide me about it. kindly guide me how can i access ICUB ears?
or what is the port name of ears of ICUB and how can i enable them?
Thanks.

randaz81, 2022-10-11:
> @randaz81 Hello I wanted to access the ear of ICUB so kindly guide me about it. kindly guide me how can i access ICUB ears? or what is the port name of ears of ICUB and how can i enable them? Thanks.

I updated https://github.com/orgs/robotology/discussions/461#discussioncomment-3815110.