robotology/mesh-support #906: Discrepancy between Physical robot and Simulator

YANGGANG1971, 2020-01-09:
When I try to move the program to a physical robot, I find that eyeball moves the opposite direction, resulting in failure of gaze control. Is it common? Do I need to configure any settings while moving the program to a physical robot? 
(Simulator performs as well as we desired.)

pattacini, 2020-01-09:
@YANGGANG1971 could you please provide us with further information about your robot, beginning with its nickname `iCub<City><Id>`?

pattacini, 2020-01-09:
What are exactly the steps you followed for your experiment? Do you have pointers to the code you ran, can you list down the software components you launched and you interacted with?

YANGGANG1971, 2020-01-09:
> @YANGGANG1971 could you please provide us with further information about your robot, beginning with its nickname `iCub<City><Id>`?

icubosaka

YANGGANG1971, 2020-01-09:
> What are exactly the steps you followed for your experiment? Do you have pointers to the code you ran, can you list down the software components you launched and you interacted with?

On the physical robot, the majority seems good. Only the robot moves towards the opposite direction as we desired. 
I uploaded all the codes, relative settings, and experimental procedures blow. 
https://www.dropbox.com/sh/gmwnzybxh0vi7tk/AAAXcAh7Io6rYBJxSPSNW2t6a?dl=0
/usr/local/src/robot/Codes/test/Recording2.doc

pattacini, 2020-01-09:
Thanks @YANGGANG1971 for the information.

Can you do the following test on the iCub_SIM first and then repeat it on the real iCub?

### On iCub_SIM
- Launch the simulator `iCub_SIM`.
- Launch `yarpmotorgui --robot icubSim`.
- Leave ticked on only the head part.
- Play with the sliders of the joints 4, 5 and 6 and note down how the eye bulbs move accordingly. 

### On iCub
- Launch the robot.
- Launch `yarpmotorgui`.
- Leave ticked on only the head part.
- Play with the sliders of the joints 4, 5 and 6 and note down how the eye bulbs move.

Is there correspondence of the movements among the simulator and the real robot?

YANGGANG1971, 2020-01-10:
Test outcome as follows,
on iCub_SIM
Joint 3rd: negative down, positive up; range: [-35 15] degrees
Joint 4th: negative left (iCub_SIM view), positive right (iCub_SIM view); range [-50 50] degrees 
Joint 5th: Positive convergence inside; [0-90] degrees

On iCub
Launch yarpmotorgui, errors occurred.
 
[image]

Joint 3rd: positive up; range: [0 ?] degrees, only positive value
Joint 4th: positive right (iCub_SIM view); range [0 ?] degrees only positive value
Joint 5th: Positive convergence inside; [0-90] degrees

[image]

pattacini, 2020-01-10:
Thanks @YANGGANG1971 for reporting the test results.

It seems that there's a more critical problem on the real iCub as the ranges are not correctly retrieved. Give us some time to figure out what might be causing this flaw.

pattacini, 2020-01-10:
Just to get around the `yarpmotorgui` and watch the face of those joint limits, could you do the following?

1. Launch the real iCub.
1. Run: `simpleClient --robot icub --part head`
1. Type: `get llim n`, where `n` accounts for the joint number, i.e. 0, 1, 2, 3, 4, 5.

What results did you get?

YANGGANG1971, 2020-01-15:
The test results are as follows,
$ get llim 0
>> Bottle get lim 0
>>llim: limits: (-30.02 22.02)
$ get llim 1
>> Bottle get lim 1
>> llim: limits: (-20 20)
$ get llim 2
>> Bottle get lim 2
>> llim: limits: (-44.97 44.97)
$ get llim 3
>> Bottle get lim 3
>> llim: limits: (-30.03 30.01)
$ get llim 4
>> Bottle get lim 4
>> llim: limits: (-30.00 30.00)
$ get llim 5
>> Bottle get lim 5
>> llim: limits: (0 50)

pattacini, 2020-01-15:
This sounds actually very strange since `yarpmotorgui` and `simpleClient` do rely on the same YARP interfaces and, from your last report, it appears that range limits are correctly set and retrieved on the real robot.

Can you provide the following further info?
1. The output of the command `yarp version` both on the PC104 (the robot hub located in the head) and on the computer where you launched the `simpleClient` and `yarpmotorgui` services.
1. The output of the commands `which simpleClient` and `which yarpmotorgui` on the computer where you launched these services.

YANGGANG1971, 2020-01-16:
On the user computer, 
> $ yarp version 
> >> YARP version 3.1.0+29-20180903.4+git131bb8c 
> $ which simpleClient 
> >> /usr/local/src/robot/install/bin/simpleClient 
> $ which yarpmotorgui 
> >>  /usr/local/src/robot/install/bin/yarpmotorgui 

Sorry, I have no idea how to operate pc104. The pc104 was put in a big box and it seems that users are hard to operate it.

pattacini, 2020-01-16:
You should be able to do:
```sh
$ ping pc104
$ ssh pc104
```

pattacini, 2020-01-20:
We have observed a similar behavior on the robot when an embedded control board stopped working. Power cycling solved the problem.

I suspect this issue is quite related to this. This would also explain why `simpleClient` was successful to read the joint bounds: it's very likely that after a reboot you got the iCub on shape again.

pattacini, 2020-01-25:
Closing, feel free to open it again if needed.

closed 2020-01-25 by pattacini