robotology/mesh-support #907: Purple iCub proximal joint cables broken for both thumbs

marco-monforte, 2020-01-15:
## Brief description of the request/failure
The proximal joint cable is broken for both thumbs on the purple iCub.
The right one broke before Christmas during control/grasping experiments for the ERC commission; the left one broke this morning, January 15th, during the initial calibration procedure.

## Accompanying material

[image]
[image]

julijenv, 2020-02-27:
Hi @marco-monforte ,
The left thumb has been restored.
It has been taken from the warehouse the following pieces to do so.
|IITCODE|Ver.|Qta |DEscription|
|--|--|--|--|
|1895||1m|carlstahl cable 0.63mm|
|2391||0.5m|carlstahl cable 0.45-0.61mm: coated|

The robot needs to be as well recalibrated, I started but I stop due to other people wanted to use it. l.rapetti and yeshavi

violadelbono, 2020-05-27:
Ciao @marco-monforte ,

We repaired the proximal joint of the right thumb. The cable was taken from warehouse:
|IITCODE|Ver.|Qta |DEscription|
|--|--|--|--|
|1815||1m|carlstahl cable 0.63mm||

We also recalibrated the robot (in .local so data will need to be pushed)

marco-monforte, 2020-05-27:
Thanks a lot, @violadelbono and @julijenv! I close the issue then.

closed 2020-05-27 by marco-monforte