robotology/mesh-support #909: R1 02 left wrist

ale-git, 2020-01-22:
## Brief description of the request/failure

Il polso sinistro non si muove e non calibra.

## Detailed context

Si sente un rumore di motore che gira a vuoto, probabilmente si è mollato un grano di una puleggia o si è rotta una cinghia.

## Accompanying material

julijenv, 2020-02-06:
hi @ale-git  e @vvasco ,
la puleggia della mano si era tolta
[image]
lo rimessa, ora potete mica guardarci se va, e chiudere la issue, grazie
cc @fbiggi

julijenv, 2020-02-07:
@ale-git, @vvasco , did you have time to check that the repair has been a success,
thx in advance

vvasco, 2020-02-07:
@julijenv I just tried to run the robot interface, but the problem is still there. 
The left wrist does not calibrate and if you open the motorgui, the left wrist tripod is like this:

[image]

julijenv, 2020-02-12:
Please @vvasco , could you attach a logfile, thx ina dvacne

vvasco, 2020-02-12:
Hi @julijenv here it is:
[robotinterface.log](https://github.com/robotology/icub-tech-support/files/4191355/robotinterface.log)

julijenv, 2020-02-12:
ciao @vvasco ,
I saw that the robot is launched with the file 
Right_Arm_No_Hand_Calibrator.xml, could you relaunch it with the normal icub_all.xml, please.
thx to let me know

vvasco, 2020-02-12:
I think that the calibration of the arms and that of the hands are just handled by different xml files, because the `yarprobotinterface.ini` points to `CER.xml`, which includes:
1. `calibrators/cer_left_arm_no_hand-calib.xml`
2. `calibrators/cer_right_arm_no_hand-calib.xml`
3. `calibrators/cer_left_hand-calib.xml`
4. `calibrators/cer_right_hand-calib.xml`

I'm not sure why it's like that (maybe @ale-git has some hints?), but if you look at the xml [here](https://github.com/robotology/cer/tree/master/app/robots/CER02/calibrators), you can notice that the only xml available are those 4 I mentioned.
So I think this is the correct configuration, even because the calibration for the right arm works.

ale-git, 2020-02-12:
Hi @vvasco, yes, it is correct.

julijenv, 2020-02-12:
everything is alright, you can finally proceed with yiur experiments.

closed 2020-02-12 by julijenv