robotology/mesh-support #910: Right hand fingers not moving

dariopasquali, 2020-01-28:
Ciao @julijenv ,
We have a problem with thumb, ring and pinky fingers of **right hand** of **iCub03**.

* The thumb correctly performs the initial calibration but, in the end, it moves to the centre of the palm and stops moving.
* Ring and pinky fingers do not move at all.

We cannot move them either with the motorgui and the arm keeps emitting a noise like it is trying to move the fingers but they do not move.

is there something we can do? Otherwise, we would need your help.

Thank you,
Dario

julijenv, 2020-01-28:
hard to say like that @dariopasquali , the cable of either thumb oppose or thumb proximal might be broken. For the last 2 joints I guess the value  is out of boundary, but it can be seen if you send me the log file of the calibration.
Please remind you it is the upmost imprtance that you join the log file to any of your porblem in order to speed the process - or else I need to request it ahahhah.
let me know

dariopasquali, 2020-02-07:
Hi @julijenv you are right!
I attach:
* a zip with the yarplog of yarprobotinterface (both the log session and the txt file, I don't know which one you prefer) [yarprobotinterface_log.zip](https://github.com/robotology/icub-tech-support/files/4169718/yarprobotinterface_log.zip)
* a picture of the right thumb

Francesco told me that you will be here next week for the iCub04 training. Maybe, if you have time, you could help us fix this.

[image]

Thank you,
Dario

julijenv, 2020-02-07:
Hi @dariopasquali , yes I will be there next Tuesday for the PhD course. Afterwards I will check youur robot S/N:020 (the old red).
I have also to make a jump in the other side of the building.
On the picture I cant really see any big problem with the mechanical cables, could you please check if the FC pcb boards is well aligned above the MAIS board (on top of the knuckles to make myself clear)
[image]

julijenv, 2020-02-14:
the thumb motor proximal has been found with a shaft broken from the gearbox output.
They are waiting to get the replacement place

closed 2020-04-23 by stale[bot]