robotology/mesh-support #911: iCubSheffield01 CAN1 for ETH 10.0.1.3 not starting properly

dcam0050, 2020-01-28:
## Brief description of the request/failure
Ran `yarprobotinterface` with icub repositories at v1.12.1 and yarp v3.1.2. 
Encountered an issue with CAN Discovery for board 10.0.1.3 - right_arm-eb3-j0_3 : `detected 4 missing eobrd_foc boards in CAN1`

## Detailed context
Here's the log from `yarprobotinterface`:

```[INFO]EthResource::askBoardVersion() found BOARD right_arm-eb3-j0_3 @ IP 10.0.1.3 of type ems4 with FW = ver 3.25 built on 2018 Jul 30 15:49 
[DEBUG] from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 10s 526m 450u: (code 0x05000015, par16 0x4000 par64 0xffffffffffff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[INFO] from BOARD 10.0.1.3 (right_arm-eb3-j0_3) @ 10s 526m 604u: CAN discovery has started for 4 eobrd_foc boards on (can1map, can2map) = (0x001e, 0x0000) with target can protocol ver 1.6 and application ver 0.0.0.
[ERROR]theNVmanager::Impl::wait() had a timeout for BOARD right_arm-eb3-j0_3 IP 10.0.1.3 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult 
[ERROR]theNVmanager::Impl::command() fails a wait() from IP 10.0.1.3 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult 
[DEBUG] from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 11s 26m 467u: (code 0x05000015, par16 0x4000 par64 0xffffffffffff0000) -> CFG: EOtheEncoderReader can be correctly configured. p16&0xf000: number of joint; primary encs: failure mask in p16&0x000f and errorcodes in p64&0x0000ffff; secondary encs: failure mask in p16&0x00f0 and errorcodes in p64&0xffff0000 + .
[ERROR] from BOARD 10.0.1.3 (right_arm-eb3-j0_3) @ 11s 26m 626u: CAN discovery after 500 ms has detected 4 missing eobrd_foc boards in CAN1:
[ERROR]1 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:1
[ERROR]2 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:2
[ERROR]3 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:3
[ERROR]4 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:4
[ERROR] from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 11s 26m 749u: (code 0x0500000e, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController cannot be configured. can discovery of 2foc boards fails. see CANdiscovery messages for more details + .
[[ERROR] from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 11s 26m 869u: (code 0x0500000e, par16 0x0000 par64 0x0000000000000000) -> CFG: EOtheMotionController cannot be configured. can discovery of 2foc boards fails. see CANdiscovery messages for more details + .
ERROR]embObjMotionControl::open() has an error in call of ethResources::serviceVerifyActivate() for BOARD right_arm-eb3-j0_3 (IP 10.0.1.3)  
[ERROR] from BOARD 10.0.1.3 (right_arm-eb3-j0_3) @ 11s 27m 16u: CAN discovery after 500 ms has detected 4 missing eobrd_foc boards in CAN1:
[ERROR]1 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:1
[ERROR]2 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:2
[ERROR]3 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:3
[ERROR]4 of 4: missing eobrd_foc BOARD 10.0.1.3:CAN1:4
```
Any ideas on how to proceed? Thanks!

## Accompanying material

dcam0050, 2020-01-28:
I've also opened the back to have a look at the LED status for the boards of the right hand and compare them with those on the left.

Here's a video for the leds on the right (the faulty one) : https://drive.google.com/file/d/1DC7FgklJ7PA84KKDums5kX__4e6ezpEh/view?usp=sharing

Here's the video for the leds on the left (the working one): https://drive.google.com/file/d/1DDzPj_Upxs2lPj-qACzSKPk0bLcEntwJ/view?usp=sharing

julijenv, 2020-01-28:
Hi @dcam0050 ,
First of all, please double check with the Firmware Updater (ETH -> discover, select EMS 10.0.1.3 ->click on go to "maintenance mode" -> clikc on discover, you should be able to see underneath the EMS 3 the CAN board attached to it, make  a printscreen if you cant see any 2FOC boards.)
You can double check with the board EMS 10.0.1.1 (the same board for the left arm) if 2FOC board are discoverable,
let me know

dcam0050, 2020-01-28:
I have put all ETH devices into maintenance mode and pressed discover.

All 2FOC boards for CAN1 for 10.0.1.3 are not discovered but I can see CAN2. Both CANs 1 and 2 are however present in 10.0.0.1. Screenshot to follow

dcam0050, 2020-01-28:
[image]

julijenv, 2020-01-28:
Ok so it was may guess.
Please check the cable going from EMS3 and the 2 top board 2FOC the contact :
There are 3 boards in total at the level of the right back of icub (2 2FOC boards 2x2=4, thats why you can see 4 foc board detected by EMS1 (10.0.1.1) on can1.
The EMS3 is beneath the lower 2FOC board.
if you follow the following diagrams:
[image]
[image]
the line is 3 wires green, black and white white labelled 3N1, check if one is broek nor damaged - to do so you may need to just remove the 4 plastic screws holding the 2FOc above the EMS and removing afterwards the standoff linking the EMS to the robot. LAter you can unattache this cable and check it. You may also try to switch the 2 piece of cable 3N1 to detect if one is working or not.
let me know

pablin992, 2020-02-19:
Continuity was measured in both cables. Both are working normally

julijenv, 2020-02-20:
Hi @pablin992 and @dcam0050 ,
Could you please just turn the motors ON and film another video of both shoulders (in one video instead of 2).
I am very interested in the boot sequence to tell you the truth. and of course, I'd like to see both shoulders in order to compare.
From the video, it looks like both 2FOC boards from the right shoulder appeared to make the exact same behaviour so it would be unfortunate that the 2FOC boards have an issue.
try also to swap the 2 can line from the EMS beneath the lower  2FOC board to understand if the can line 2 of the EMS could have an issue.
[image]
I am hanging for a short reply to fix this issue quick to ake your robot up and running again so please, don't make me longing too much for your reply, cheers ;)

dcam0050, 2020-02-25:
Here's a video of the boot sequence of the left and right 2FOC Boards

https://youtu.be/OfWNntGspng

We will be trying the CAN line swap soon

julijenv, 2020-02-25:
Hi @dcam0050 ,
the video is awesome because we can see that every 2FOC board in the back behave similarly so it is a good indicator that most probably the problem is link to the talking to CAN line instead of problem with the board themself. It may be a problem from the EMS CAN lines, please do the swap and tell me if you can see 2FOC boards with FirmwareUpdate.
Let me know,
Cheers

dcam0050, 2020-02-25:
Hi @julijenv, swapping the CAN lines worked well! We've opened another issue to address further problems we have with booting iCubSheffield.

julijenv, 2020-02-25:
Hi @dcam0050,
You closed the issue but I dont think it is actually finished or I am wrong, does the robot start???
unfortunately swappping the CAN line just make us learn that probably the EMS board has one of its CAN line broken and if it is the case You need to buy another one to restore the normal use of the robot.
let me know

julijenv, 2020-02-27:
Hi @dcam0050 , @pablin992 and @Quiescitus ,
it would be nice to get in touch on skype to discuss and sum up a bit the problems you are facing #928 .
my skype ID is icub.support,
let me know

closed 2020-07-05 by stale[bot]