robotology/mesh-support #912: iTeenGenova01: Electric issues

valegagge, 2020-01-28:
Here I report two failures:
 **1.** Yesterday I launched `yarprobotinterface` several times and in some case I ran into following error:
```
from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 54s 257m 472u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
Since the error didn't occurred always, I think it could be due to a riuned crimp of elbow encoder. 

**2.**
While I was running `yarprobotinterface` the board with IP address 10.0.1.27 stoped to transmit.
I ran again `yarprobotinterface` but I din't get such error. Could you verify the ethernet connection **to** board with IP address 10.0.1.27?

thank you

cc @vtikha

marcellosavoldi, 2020-01-28:
@julijenv , did you note my note on the left arm board " wire brocken" ?
There was a white wire broken....

julijenv, 2020-01-28:
@marcello, the broken wire you mention was repaired and is not responsible for this prblem (the broken wire was on the right arm :D).
@valegagge, I checked and I found something but I needed to test it and it seems that the yarprobotinterface and the configuration files has been changed (the yarplogger quote a lot of problem related to V6 needed) - so I guess I would need you to double check if I found the cuprit or not.
As for the board 27, could you please let me know the error message your received??1 to know if it was from the I/int or O signal.
C U tomorroz

julijenv, 2020-01-29:
the problem with the left _arm has been setteld thx to the help of @ale-git .
Remains the problem with right forearm, please give me feedback about the failure, please
I mean after that you realize that the IP 27 was not responding, could you be able to see or detecteing the board beneath (I mean the board that followed it on the ETH line).
let me know

vtikha, 2020-01-29:
Ciao @julijenv I did a few tests this morning and I report some of the findings.

The first time you run the `yarprobotinterface` after turning on the motors there are **no issues** or errors being reported on the printouts while calibrating and also while closing/parking. 

Having said that, once you restart `yarprobotinterface` (without turning off the motors) there are a lot of **errors** being reported on multiple boards, see below. The behavior on the robot is somewhat strange as, sometimes, the calibration seems to have been executed properly  and other times, some parts fail (left arm / left forearm etc...). 

``` bash
Left_Arm_Calibrator : joint  3  has an hardware fault, skipping park 

[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 85s 649m 682u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

``` bash
Left_Arm_Calibrator : joint  3  has an hardware fault, skipping park 

[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 85s 649m 682u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

``` bash
[ERROR]Right_Arm_Calibrator : detected an hardware fault on joint  9  before starting the calibration procedure! 
[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  10  before starting the calibration procedure!
[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  11  before starting the calibration procedure! 
[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  12  before starting the calibration procedure! [
ERROR[ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  13  before starting the calibration procedure! 
]Right_Arm_Calibrator : detected an hardware fault on joint  10  before starting the calibration procedure! 
[[ERROR]Right_Arm_Calibrator : detected an hardware fault on joint  11  before starting the calibration procedure! 
ERROR]Left_Arm_Calibrator : detected an hardware fault on joint  14  before starting the calibration procedure! 
[ERROR[ERROR]Right_Arm_Calibrator : detected an hardware fault on joint  12  before starting the calibration procedure! 
]Left_Arm_Calibrator : detected an hardware fault on joint  15  before starting the calibration procedure! 
[[ERRORDEBUG]Right_Arm_Calibrator : detected an hardware fault on joint  13  before starting the calibration procedure! 
]Left_Arm_Calibrator : Joints calibration order:  
[ERROR]Right_Arm_Calibrator : detected an hardware fault on joint  14  before starting the calibration procedure! 
[ERROR]Right_Arm_Calibrator : detected an hardware fault on joint  15  before starting the calibration procedure!

[ERROR]Left_Arm_Calibrator : joint  13  has an hardware fault, skipping park
```

For validation purposes I turned off and on the motors once again and all seemed ok until  this happened again

``` bash
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 210s 861m 974u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

and calibration of the **left elbow failed**

please let m know when you can have a look at it so that I can be there to try things with you. 
Thanks

vtikha, 2020-01-30:
Ciao @julijenv , after your intervention on the left arm, I have tested the system 4-5 times in various configurations and all is working well. Thank you.
The hardware fault errors are coming from the fingers but it seems just to be a message from the firmware as the hand is fully controllable. I think the best thing would be to open another issue for this error on the firmware side.

valegagge, 2020-01-30:
Hi @julijenv ,
unfortunately the issue on 2foc encoder comes back!!! :(

```
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 530s 828m 13u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```
Also together with the same issue on torso joint number 0!!

```
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 531s 564m 327u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

So I think the issue about "the big cable" supply was not related to 2foc encoder issue.

julijenv, 2020-01-30:
Ciao @valegagge , mi sembra un problema piu al livello FW, @ale-git potresti darci un occhiata, spunta un po casualmente errore sempre di 2FOC su canali var. Mai gli stessi a seconod dell'accensione.
per il gomito ho riuscito risolvere mettendo il braccio a start-up in posizione diversi di quellla di riposo ma neanche quella di calibrazione.

closed 2020-04-26 by stale[bot]