robotology/mesh-support #913: Right arm middle finger distal joint stopped to work

frank-foerster, 2020-01-30:
## Brief description of the failure
The distal joint of the right arm middle finger (joint #13 in yarpmotorgui) of iCubHertfordshire has stopped to work.
I can hear a funny sound in or around the hand which lets me speculate that the cable might have snapped. 

## Question
How do I figure out

1. whether the cable has indeed  snapped?
2. If (1) is true, which cable to I need to replace the snapped one?
3. If (1) is true, do we have any instructions on how to rewire the finger?

Many thanks for your help.

julijenv, 2020-01-31:
Hi @frank-foerster ,
1. To check if the cable is broken, is indeed quite simple. Please spin the screw following piece inside the forearm of your robot
[image]
by spinning it in the direction daw you should make the middle distal closed, if after 2-3 turns nothin happen, you can be certain that the cable is broken.
2. please use cable (that probably we have sent to you - IITCODE 2391 - 0.61mm outer diameter (it is coated) core 0.45mm), take 40cm of it
3. This is an on going project, unfortunately as for now ,nothing still ready to be published.
therefore I can just gives you teaser
[image]
[image]

[image]

[image]
hope this will be from any help

marcoaccame, 2020-01-31:
Hi @julijenv, 

> 3. This is an on going project, unfortunately as for now ,nothing still ready to be published.
therefore I can just gives you teaser

it is true that it is an ongoing project. Because documentation can always be improved 😇.

But it is also true that the site has gone public and, even if it its content is still WIP, I am positive about pointing users to the correct URL: [iCub Cabling Manual: the hand](https://icub-tech-iit.github.io/documentation/mkdocs/tendons_icub/site/lower_arm_V1/)

frank-foerster, 2020-01-31:
Hi @julijenv and @marcoaccame , 
thanks for the quick reply. 
I just tried turning the screw and the cable is broken indeed. 

Awesome, that there is documentation now! 
I'll try to follow the instructions there and if I get stuck, I will get back to you!

frank-foerster, 2020-02-06:
Hi @julijenv ,
only know I got to start the repair, but removed the ripped cable and think I know how to do the repair. But in terms of the glue you are using to fix the knots: is that just regular superglue? 

Thanks for your help

julijenv, 2020-02-07:
hi @frank-foerster , yes it is, just wait until it dries out or just use a tissue to remove the excess.
cheers

frank-foerster, 2020-02-07:
Hi @julijenv ,
thanks for the quick feedback. Two  - hopefully last - questions:

1. The 40 cm cable will be much too long - the one I took out (the one that ripped) was much shorter than that. It states in the description that I should make the knot close to the fingernail 1 to 2 cms  away from the nail. I guess the idea is that I then, once I made the final know, manually roll up the rest onto the "winch" of the motor where the cable is rolled on and off by manually turning the motor until the cable is more or less "under tension". Is that correct? 

2. Once I did that, do I need to recalibrate the (control software for) the fingers? (with the Fingers PID tuner or the like?). I'm asking because I have never done this before.

Thanks again.

julijenv, 2020-02-07:
Hi @frank-foerster ,
1. 40cm is long enough and it is not too short, remember you need to make 2 knots on both sides and so by this, part of the cable will be removed, furthermore, you need to have enough spare cable at the level of the fingertip in order to make the knot.
2. you do not need to recalibrate anything, just spin the motor as shown in the picture above to start closing the finger. just to make sure that the motor goes in the right direction.

frank-foerster, 2020-02-07:
It's done! Just tried my control script and it all seems to work well again. 
And it was easier than expected! 
Thanks for you help!
And thanks for creating repair guidelines - they should make the requests for help a little bit less frequent / likely.

closed 2020-02-07 by frank-foerster