robotology/mesh-support #915: A little piece of plastic fell from Green iCub

HosameldinMohamed, 2020-02-03:
## Brief description of the request/failure
We were handling the Green iCub (iCubGenova04) during an experiment on Friday 31/01/2020 where a small plastic piece fell from the robot (see the photo). I am not sure if it got broken or it was glued.

## Detailed context
The piece fell from inside the cover of the robot. I am not sure if it fell from the front part of the back part.

I put the piece on my desk.

## Accompanying material

[image]

julijenv, 2020-02-04:
@HosameldinMohamed , this  piece is part of the chest holder, it has been broekn and reglued and repaired 1000 times of times, every time we put a new one it got broken right after , so I will reglue it again. Dont worry

HosameldinMohamed, 2020-02-04:
> @HosameldinMohamed , this piece is part of the chest holder, it has been broekn and reglued and repaired 1000 times of times, every time we put a new one it got broken right after , so I will reglue it again. Dont worry

Thank you @julijenv :+1:

Elio75, 2020-02-12:
**ok the new piece has been replaced ...**
@julijenv @Mick3Lozzo  Gesino se stai modificando il pezzo in questione dai una occhiata ai fori delle viti d'attacco , c'è poco materiale e penso che basta poco in chiusura per rompersi... (guarda imagine)
[image]

potresti magari attaccare il pezzo qua 
[image]

julijenv, 2020-02-12:
Please @Mick3Lozzo , as you are making the new piece, try to use the feedback given by @Elio75 , thx

closed 2020-02-12 by julijenv