robotology/mesh-support #918: Unable to move thumb proximal joint

fbottarel, 2020-02-07:
## Brief description of the request/failure
The proximal joint of the right thumb does not move either with yarpmotorgui nor by hand. 

## Detailed context
Setting the joint to idle does not change anything. Currently, the joint is frozen close to the palm.

## Accompanying material

fbottarel, 2020-02-07:
@xEnVrE

julijenv, 2020-02-07:
it has been checked the thumb proximal is broken - I need at least 2-3hours to restore it,
I will seee next week to make it done.
So far, @xEnVrE @damianomal told me that they dont use the grasping recently, so I will do it later.

julijenv, 2020-02-21:
it has been restored the thumb proximal cable.
|IITCODE|Qta|Description|
|---|---|---|
|1895|1m|carlstahl 0.63mm|
thx to @xEnVrE , the commit has been put in robots-configuration correctly

closed 2020-02-21 by julijenv