robotology/mesh-support #920: Left arm CAN problem

S-Dafarra, 2020-02-11:
When running the ``yarprobotinterface``, the logger is filled with messages like 
```
173062	485.527100	ERROR	 from BOARD 10.0.1.24 (left_arm-eb24-j4_7), src CAN1, adr 0, time 53s 466m 108u: (code 0x00000019, par16 0x0b01 par64 0x0000000000000000) -> SYS: EOtheCANservice could not tx frames on CAN bus. In par16 there is: on msb the size of txfifo, on lsb a code. + .
```

In addition, the left wrist pronosupination moves much slower than usual.

cc @GiulioRomualdi @MiladShafiee @julijenv

julijenv, 2020-02-27:
Hi @S-Dafarra ,
Could you please send an entire log-file and not only the first ERROR message displayed you found.
unfortunately, the ERROR message that you quoted is related to skin patches-boards and not to controlling board itself.
please add the absolute entire log=file when next time the pronosuppination don't work,\thx in advance

S-Dafarra, 2020-02-27:
Hi @julijenv, you are absolutely right. Indeed, the problem was that the logger was filled with basically only that message and quickly saturate, hence we were not able to see any new error. 

How can we mute that error?

julijenv, 2020-02-27:
Hi @S-Dafarra , just remove the skin from the left arm inside  icub_all-wbd.xml

S-Dafarra, 2020-04-28:
I would keep it open for the time being.

closed 2020-07-05 by stale[bot]