robotology/mesh-support #922: iCub03 torso yaw cable broken

dariopasquali, 2020-02-12:
CIao @julijenv ,

We just broke the yaw cable of the torso (attached pictures).
It happened during a normal startup, no weird movements.

The problem is that me, Giulia and Elena have to make experiments in this and the next week.
So few questions:
* We never replaced that cable, and we don't have a spare one. is it simple enough to repair it by myself if I came to take the cable? Can you teach me?
* Otherwise, could you come here asap (like tomorrow) and fix it?

Thank you,
Dario

[image]
[image]
[image]

julijenv, 2020-02-12:
Ciao @dariopasquali,
Here is the link to get the documentation on how to repair iCub
https://icub-tech-iit.github.io/documentation/mkdocs/tendons_icub/site/waist_legs/
You need cable 13 to proceed with this guide.
Or else I pass by tomorrow morning before going to IIT sanqui

dariopasquali, 2020-02-12:
Ciao @julijenv ,
thank you!
If you can come here tomorrow morning would be perfect! This way I can also learn directly from you.
if you are not in sanqui we can go there and take the cable.

Let me know.
Thank you again.
Dario

pattacini, 2020-02-12:
@dariopasquali @julijenv I'm assuming this is `iCubGenova03`.
Amend the title as well as the label if this info is not correct.

pattacini, 2020-02-12:
@dariopasquali I've just invited to join Robotology org as a member 😉

julijenv, 2020-02-14:
it has been taken from the warehouse

|IITCODE |Qta |DEscription|
|:--:|:--:|:--:|
|4770.1 |1 |WAIST TOP CABLE 13|

closed 2020-02-14 by julijenv