robotology/mesh-support #924: [iCubGenova01] Restore original fingertips and top cover of left hand

xEnVrE, 2020-02-21:
## Brief description of the request
After a f2f discussion with @pattacini and considering that the VVV20 school is approaching, I think we should restore the original fingertips and the top cover of the left hand of `iCubGenova01`.

As regards the fingertips, even if some of them might be working (we did not test all of them), they were mounted temporarily and the wire connecting them are exposed and can be easily damaged during an experiment.

Restoring the original cover is very easy and I can also take care of it if the proper screwdriver is available.

cc @pattacini @lornat75 @julijenv @simeonedussoni

pattacini, 2020-02-21:
@simeonedussoni @julijenv can we plan some time next week to restore the old functionalities?
We shall be quick as Blacky will be leaving to attend VVV20 quite soon.

julijenv, 2020-02-21:
it is quite simple and fast @pa, @xEnVrE  1 hour max

xEnVrE, 2020-02-24:
Closing as both the cover and the fingertips have been restored.

Thanks @julijenv.

closed 2020-02-24 by xEnVrE

julijenv, 2020-02-25:
the fingertips were replaced and tested with @xEnVrE , they look they work alright.