robotology/mesh-support #925: Connection between the iCub and PSU

astar-wuy, 2020-02-24:
There seems to be a missing component between the PC104 part of the connection and the PSU. Please kindly help. Thank you.

[image]

julijenv, 2020-02-25:
Hi @astar-wuy ,
Actually this hanging cable is a one pin of  a Schottky diode, I presume the other part of it is still attached to the back of the power supply. Buy a new one and resolder the pins  and you are done.
Here the diode information:

4438 | STPS15L25D | Diodes, zener, transzorbs, TVS | Diode schottky,   Vrrm: 25V, If: 15A, Ifms: 250A, THD, TO220
-- | --  | -- | --

Here 2 references

 for the diode you need to buy

St microelectronics | STPS15L25D | FARNELL ITALIA S.R.L. | 9803408
-- | -- | -- | --
  |   | RS COMPONENTS S.P.A. | 714-6885

Here a diagram on how to connect the diode

[image]

[image]

have a nice day.

astar-wuy, 2020-02-26:
Hi  Julien,
  Thanks for the drawings. Attached are the done wiring. Please let us know if anything is done wrongly. Thanks.

Best wishes,
Yan

[Diode_Schottky]<https://user-images.githubusercontent.com/7140395/75243867-fb56c700-57ca-11ea-9164-bbf32fdd3d83.PNG>

[image]<https://user-images.githubusercontent.com/7140395/75243549-5e942980-57ca-11ea-86a0-03df16c701c3.png>

This e-mail and any attachments are only for the use of the intended recipient and may contain material that is confidential, privileged and/or protected by the Official Secrets Act. If you are not the intended recipient, please delete it or notify the sender immediately. Please do not copy or use it for any purpose or disclose the contents to any other person.

astar-wuy, 2020-02-26:
[image]
[image]

julijenv, 2020-02-26:
Hi @astar-wuy ,
almost perfect! The only thing that was not written so that why I need to tell you about is:
You need to place a thermally conductive BUT electrically isolated tape in between the back of the diode and the power supply.
here the information at this regards:

 | RS COMPONENTS S.P.A. | 477-4856| Thermally conductive double sided adhesive tape Bond-Ply 100, 152,4x152,4mm, Th 0,127mm, T 30/120°C, 0,8W/m•K|Bergquist | BP105-0.005-00-00-125X1506 |
|-- | -- | -- | --| --|

Hope this will help
let me know

astar-wuy, 2020-03-03:
Hi Julien,
  Thanks a lot! We have got the tape and the robot is now up and running. Thanks once again!

Best wishes,
Yan

closed 2020-03-03 by julijenv