robotology/mesh-support #927: Green icub left shoulder cable broken and right shoulder going to HW fault often

Yeshasvitvs, 2020-02-24:
## Brief description of the request/failure

While performing retargering experiments, the left shoulder cable of the green robot is broke. Also, the right shoulder joints are going to HW fault often, possible due to some loose connection. @julijenv can you please take a loot at it tomorrow morning. Thanks.

CC @lrapetti @kouroshD 

## Detailed context

## Accompanying material

pattacini, 2020-02-24:
@Yeshasvitvs does your retargeting mechanism take into account the constraints on the cable lengths that in turn impact the movements of the three shoulder joints?

Just wondering as the lack of such a safety guard often results in breaking the shoulder cables or having those cables that snap out of their places. 

Here's a quite old [reference][1] describing these limitations.

[1]: https://wiki.icub.org/wiki/Shoulder_range_of_movements

lrapetti, 2020-02-24:
> @Yeshasvitvs does your retargeting mechanism take into account the constraints on the cable lengths that in turn impact the movements of the three shoulder joints?

@pattacini 
yes, they should be taken into account, see https://github.com/robotology/human-dynamics-estimation/issues/150 and https://github.com/robotology/human-dynamics-estimation/blob/master/conf/xml/RobotStateProvider_iCub.xml#L79

pattacini, 2020-02-24:
Good boys 👍

julijenv, 2020-02-25:
The robot integrity has been restored, please test it by launch the yarprobotinterface and close the issue thx in advance @Yeshasvitvs @gabrielenava

Yeshasvitvs, 2020-02-25:
Everything seems fine. Thanks @julijenv 
Closing.

closed 2020-02-25 by Yeshasvitvs

julijenv, 2020-02-27:
It has been taken from the warehouse the following pieces to restore the integrity of the iCub.
|IITCODE|Ver.|Qta |DEscription|
|--|--|--|--|
|4764|2|1|Shoulder rotation cable 7|