robotology/mesh-support #928: iCubSheffield 01 - AEA Encoder Invalid Data

Quiescitus, 2020-02-25:
## Brief description of the request/failure
When booting up the iCub with `yarprobotinterface` an error is encountered on Board 10.0.1.5 with the magnetic sensor.

Here's the startup log:

```||| configuring
||| default config file specified as yarprobotinterface.ini
||| checking [/usr/local/src/robot/icub-main/build-pc104/yarprobotinterface.ini] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/yarprobotinterface.ini] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/yarprobotinterface.ini
||| finding file [config]
||| checking [/usr/local/src/robot/icub-main/build-pc104/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/./icub_all.xml
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/./icub_all.xml 
[DEBUG]yarprobotinterface: using xml parser for DTD v3.x 
[DEBUG]Reading file /usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/./icub_all.xml 
[DEBUG]Preprocessor complete in:  0.0641332 s 
||| finding paths [plugins]
||| checking [/usr/local/src/robot/icub-main/build-pc104/plugins] (pwd)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubSheffield01/plugins] (robot)
||| checking [/home/icub/.config/yarp/plugins] (YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/plugins] (YARP_DATA_HOME)
||| checking [/etc/yarp/plugins] (YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/yarp/build-pc104/share/yarp/plugins
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins] (YARP_DATA_DIRS)
||| found /usr/local/src/robot/iCubContrib/share/ICUBcontrib/plugins
||| checking [/usr/local/src/robot/icub-tests/suits/plugins] (YARP_DATA_DIRS)
yarp: Port /icub/yarprobotinterface active at tcp://10.0.0.2:10002/
[INFO]startup phase starting... 
||| finding paths [plugins]
[INFO]***** Configuring cartesian controller *****
[INFO]Acquiring options for group GENERAL...
[INFO]Acquiring options for group DRIVER_0...
[INFO]Acquiring options for group DRIVER_1...
[INFO]PLANT_MODEL group detected
[INFO]DH Table:  (H0 (0.0 -1.0 0.0 0.0 0.0 0.0 -1.0 0.0 1.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0)) (HN (1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0)) (link_0 (A 0.032) (D 0.0) (alpha 90.0) (offset 0.0) (min -22.0) (max 84.0) (blocked 0.0)) (link_1 (A 0.0) (D -0.0055) (alpha 90.0) (offset -90.0) (min -39.0) (max 39.0) (blocked 0.0)) (link_2 (A 0.0233647) (D -0.1433) (alpha -90.0) (offset 105.0) (min -59.0) (max 59.0) (blocked 0.0)) (link_3 (A 0.0) (D 0.10774) (alpha -90.0) (offset 90.0) (min -95.5) (max 5.0)) (link_4 (A 0.0) (D 0.0) (alpha 90.0) (offset -90.0) (min 0.0) (max 160.8)) (link_5 (A 0.015) (D 0.15228) (alpha -90.0) (offset 75.0) (min -37.0) (max 100.0)) (link_6 (A -0.015) (D 0.0) (alpha 90.0) (offset 0.0) (min 5.5) (max 106.0)) (link_7 (A 0.0) (D 0.1413) (alpha 90.0) (offset -90.0) (min -50.0) (max 50.0)) (link_8 (A 0.0) (D 0.0) (alpha 90.0) (offset 90.0) (min -65.0) (max 10.0)) (link_9 (A 0.0625) (D -0.02598) (alpha 0.0) (offset 0.0) (min -25.0) (max 25.0)) (numLinks 10) (type left_v2) 
yarp: Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/in active at tcp://10.0.0.2:10003/
yarp: Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/out active at tcp://10.0.0.2:10004/
yarp: Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/rpc active at tcp://10.0.0.2:10005/
yarp: Port /icub/cartesianController/left_arm/command:i active at tcp://10.0.0.2:10006/
yarp: Port /icub/cartesianController/left_arm/state:o active at tcp://10.0.0.2:10007/
yarp: Port /icub/cartesianController/left_arm/events:o active at tcp://10.0.0.2:10008/
yarp: Port /icub/cartesianController/left_arm/rpc:i active at tcp://10.0.0.2:10009/
[INFO]created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
[INFO]***** Configuring cartesian controller *****
[INFO]Acquiring options for group GENERAL...
[INFO]Acquiring options for group DRIVER_0...
[INFO]Acquiring options for group DRIVER_1...
[INFO]PLANT_MODEL group detected
[INFO]DH Table:  (H0 (0.0 -1.0 0.0 0.0 0.0 0.0 -1.0 0.0 1.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0)) (HN (1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0)) (link_0 (A 0.032) (D 0.0) (alpha 90.0) (offset 0.0) (min -22.0) (max 84.0) (blocked 0.0)) (link_1 (A 0.0) (D -0.0055) (alpha 90.0) (offset -90.0) (min -39.0) (max 39.0) (blocked 0.0)) (link_2 (A -0.0233647) (D -0.1433) (alpha 90.0) (offset -105.0) (min -59.0) (max 59.0) (blocked 0.0)) (link_3 (A 0.0) (D -0.10774) (alpha 90.0) (offset -90.0) (min -95.5) (max 5.0)) (link_4 (A 0.0) (D 0.0) (alpha -90.0) (offset -90.0) (min 0.0) (max 160.8)) (link_5 (A -0.015) (D -0.15228) (alpha -90.0) (offset -105.0) (min -37.0) (max 100.0)) (link_6 (A 0.015) (D 0.0) (alpha 90.0) (offset 0.0) (min 5.5) (max 106.0)) (link_7 (A 0.0) (D -0.1413) (alpha 90.0) (offset -90.0) (min -50.0) (max 50.0)) (link_8 (A 0.0) (D 0.0) (alpha 90.0) (offset 90.0) (min -65.0) (max 10.0)) (link_9 (A 0.0625) (D 0.02598) (alpha 0.0) (offset 180.0) (min -25.0) (max 25.0)) (numLinks 10) (type right_v2) 
yarp: Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/in active at tcp://10.0.0.2:10010/
yarp: Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/out active at tcp://10.0.0.2:10011/
yarp: Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/rpc active at tcp://10.0.0.2:10012/
yarp: Port /icub/cartesianController/right_arm/command:i active at tcp://10.0.0.2:10013/
yarp: Port /icub/cartesianController/right_arm/state:o active at tcp://10.0.0.2:10014/
yarp: Port /icub/cartesianController/right_arm/events:o active at tcp://10.0.0.2:10015/
yarp: Port /icub/cartesianController/right_arm/rpc:i active at tcp://10.0.0.2:10016/
[INFO]created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
||| finding paths [plugins]
[WARNING]Rotor: Using default value = 1
[INFO]Position Pids section found, new format
[INFO]Position Pids successfully loaded
[WARNING]Torque control parameters not found for part , skipping...
[DEBUG]DEBUG parameters section NOT found, skipping...
[DEBUG]IMPEDANCE parameters section NOT found, skipping...
[WARNING]jntVelMax: Using default jntVelMax=100 deg/s
[... 1008 lines cut ...]
None
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 392s 124m 270u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[DEBUG]cfw2can [0] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.04[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]cfw2can [0] printing boards infos:
None
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 397s 125m 270u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
[ERROR]unable to complete calibration: joint 1 in 'hw_fault status' inside doneRaw() function
[... cut ...]
[ERROR]embObjMotionControl::checkRemoteControlModeStatus(2, 7103593) is unable to check the control mode of BOARD torso-eb5-j0_2 IP 10.0.1.5 because it is now in HW_FAULT
[ERROR]embObjMotionControl::checkRemoteControlModeStatus(0, 7103593) is unable to check the control mode of BOARD torso-eb5-j0_2 IP 10.0.1.5 because it is now in HW_FAULT
[ERROR]embObjMotionControl::checkRemoteControlModeStatus(1, 7103593) is unable to check the control mode of BOARD torso-eb5-j0_2 IP 10.0.1.5 because it is now in HW_FAULT
```

julijenv, 2020-02-25:
Could you open the yarpmotorgui and check which joint of the torse is not giving you back values.
look the encoder value for those 3 joints and move the robot a bit when meanwhile rotating the 3 axes holding the robot (move the robot yourself, that was my sense)
give me feedback (screenshot of the robotmotorgui encircling the joint with no encoder value changing).
thx in advance

dcam0050, 2020-02-25:
The issue is with the torso_roll encoder which doesn't change no matter in which direction I move the iCub.

Occasionally during the startup process, the robot does a small jolt forward like it's about to get up but that's it.

Also depending on how much time I leave between switching motors on and running `yarpmotorgui` I get different combinations of limbs and / or head with hardware faults.

[image]

julijenv, 2020-02-26:
Hi @dcam0050  and @Quiescitus ,
Could you please check the integrity of the cable at the level of the waist-belly.
[image]

I do believe that it got pinched - more probably - or follow the cable and see if one the wire got broken or uncrimped.
let me know

Quiescitus, 2020-02-26:
Hi @julijenv,
We followed that cable and it seems there's a section which feels like it was pinched (on image 1).

[image]
[image]
[image]

Not 100% certain, as we couldn't test with an amperemeter since one of the ends is sealed.
Any suggestions?

julijenv, 2020-02-26:
Hi @Quiescitus ,
this part looks pinched to me
[image]
remove the connector from the board and make a continuity tet with the board at the level of the belly (remove the 2 lateral screws, BEFORE just make sure you note how the plate and subsequently the sensor are placed because or else you could have a 180 degrees changed later when you will remount the overall).
Let me know

Quiescitus, 2020-02-26:
@julijenv, That's indeed the case...
The cables are damaged around that spot. Do we need to order a replacement?

julijenv, 2020-02-26:
Hi @Quiescitus ,
I will discuss tomorrow about sending you a spare part of it BUT my fear is that I will send you something that needs to be accompanied by another board from issue #911. Unfortunately, it looks like the board as a problem but you didn't quite really answer to my last question on it.
I need to see the screenshot of FirmwareUpdater when you have swapped the 2 can line and see either the FT sensor show up or not?
Let me know

violadelbono, 2020-03-03:
The following items have been taken from the warehouse:

| IIT CODE       | Rev      |  Quantity          | Description  |
| ------------- |:-------------:| -----:| -----:|
|  1810     | A |  1 | AEA-001 Electronic board, 12-bit digital absolute encoder with spi interface |
|  1310     |   | 0.3 m | Teflon insulated cable, 5 poles, AWG32, 250V, with black braid, tape and jacket  |

Quiescitus, 2020-03-04:
Thanks, we'll update this once there's more info on the order

violadelbono, 2020-03-04:
The details about the pieces serve as a record just for us.
You will have more info at the shipping phase.

closed 2020-05-10 by stale[bot]