robotology/mesh-support #929: Green icub left shoulder cable broken

lrapetti, 2020-02-26:
## Brief description of the request/failure
While performing retargering experiments, at the startup of the experiment the robot (on the pole) did an unexpected fast motion hitting with the arm and torso.
Luckily, despite scaring crack noise, at the moment it looks like only the shoulder brake, plus we found a finger piece on the ground (shown in the picture below), but not sure if it comes from this failure since all the fingers cover seems to be fine

## Accompanying material
[image]
[image]

CC @julijenv @Yeshasvitvs

lrapetti, 2020-02-27:
I tried to run the `yarprobotinterface` without the left arm and it started correctly, so no other joint is damaged.

julijenv, 2020-02-27:
it has been used the following pieces to restore the robot
|IITCODE|Ver.|Qta |DEscription|
|--|--|--|--|
|4759|1|1|Shoulder rotation cable 2|
|4764|2|1|Shoulder rotation cable 7|
|4682|A1|1|AEA cover wrist|

closed 2020-02-27 by julijenv