robotology/mesh-support #931: R1: Error on BOARD EB10 when launching yarprobotinterface

miccol, 2020-02-27:
## Brief description of the request/failure
The yarprobotinterface cannot be launched.

## Detailed context
When running the yarprobotinterface on the R1 robot we get the error in the figure below.
We tried to powerloop several times but the problem is still there

## Accompanying material

[image]

miccol, 2020-02-27:
@safoex

julijenv, 2020-02-27:
Hi @miccol , what did you do with the neck?!?!?!?

julijenv, 2020-02-27:
The MC4-PLUS board was found not correctl funxtionning, after deeper investigation it looks like the board was dead in some part.
I substitued the board and everything went back to normal.

it has been taken from the warehous ethe following piece to restore the R1 robot
|IITCODE|Ver.|Qta|description|
|--|--|--|--|
|5443|D|1|MC4-PLUs|

closed 2020-02-27 by julijenv

vtikha, 2020-02-27:
Thank you @julijenv

miccol, 2020-02-28:
Thank you for the quick fix @julijenv !