robotology/mesh-support #932: iCubGenova02 bad general loose conditions

DanielePucci, 2020-02-28:
## Brief description of the request/failure
We started working on the robot and preliminary qualitative checks clearly show that the robot is in a bad general mechanical conditions - see videos below. Unfortunately, we cannot upload videos with sounds, that would have helped understand also the general situation 

## Detailed context

## Accompanying material

[image]
[image]
[image]
[image]

CC
@julijenv @S-Dafarra @lrapetti @marco-monforte  @ChiaraBartolozzi

traversaro, 2020-02-28:
> Unfortunately, we cannot upload videos with sounds, 

Small note: you can attach arbitrary file size by compressing them in a `.zip` .

julijenv, 2020-02-28:
Hi @traversaro , @DanielePucci  and @lrapetti ,
I just finish to reense everything that I could do on the fly, before @lrapetti start using it.
the left arm needed clearly to get a tiny tensing.
I will need to get the robot for more time to be able to adjust the overall situation

violadelbono, 2020-06-09:
Hello guys,

The cables of the torso, legs and arms were tensed.
The fine calibration of the torso was corrected (there was an offset of about 1.8 deg in joint 1)

marco-monforte, 2020-06-09:
Thanks @violadelbono !

What is missing is only the head then, which is more complicated to tense.
In the next weeks, the robot is needed (at least) by Giulia D'Angelo for some experiments, so we might go back to this once the robot won't be used.

@DanielePucci @traversaro @lrapetti feel free to close the issue :+1:

lrapetti, 2020-06-09:
Great, thank you @violadelbono and @marco-monforte!

closed 2020-06-09 by lrapetti

DanielePucci, 2021-11-04:
CC @Uboldi80 @gsisinna @GiacomoVassallo13 we are having a [similar head issue](https://user-images.githubusercontent.com/5045846/75526576-562e2f80-5a12-11ea-97dd-a931c0f69d13.gif) on iCub 3, it would be nice to discuss about this before opening a specific issue for iCub 3 head