robotology/mesh-support #935: Broken eye tilt belt

loreucci, 2020-03-06:
## Brief description of the request/failure
The belt for the eye tilt broke (iCubPisa01) and we have no more spares. We tried searching for a replacement from [http://wiki.icub.org/wiki/Springs,_belts_and_cables](http://wiki.icub.org/wiki/Springs,_belts_and_cables) but it seems that the code for that specific belt is no longer available from the manufacturer website. Do you have a new code or a different supplier from which we could get the spare part?

julijenv, 2020-03-12:
Hi @loreucci ,
thx to @salvi-mattia, we found out the old correct code for this specific belt.
**53.6MXL025**
[image]
[image]

we double check and I sent a mail to sitspa our provider and ask if it is still available. It is a belt that has to be done on demand unfortunately, so I dont know exactely what will be the price for it.
let you know

julijenv, 2020-05-13:
Hi @loreucci ,
 Here what I received for the requested belt from the supplier sitspa italy
[Offer-166625.pdf](https://github.com/robotology/icub-tech-support/files/4621363/Offer-166625.pdf)
the only bad part of it , is that you need to order at least 30 pieces of it.
let me know

closed 2020-07-20 by stale[bot]