robotology/mesh-support #936: iCub power connection question

iPsych, 2020-04-17:
Dear all,

I set up two PSU and trying to connect them.
Is there any visualized guideline or instruction for connecting 48V and 12V cables?

Also I am afraid I don't have any information about DB15 cable. What does it for?

Below I attach the photos. of the cables : )

[image]
[image]

julijenv, 2020-04-20:
Dear, @iPsych ,
I see that you perfectly found the documentation for the Harness of the cable but it seems that you didnt find the one for the connection for the power supply, here it is.
[6486_ExtPowerCable_3.0.1.pdf](https://github.com/robotology/icub-tech-support/files/4504065/6486_ExtPowerCable_3.0.1.pdf)
To be faster \:
both yellow ring labelled *12V* are connected to the positive pin of the 12V power Supply and the black one to GND (negative pin of the 12V power supply)
[image]
Th 2 red ring labelled *48V* are connected to the positive pin of the 48V power Supply and the 2 black ones to GND (negative pin of the48V power supply)
[image]
Moreover you have a black wire (with the Heart symbol) which connect both GND of the 2 power supplies.
THe DB15 needs to be connected in the bak of the power supply 48V
[image]

Hope it is clear, let us know

iPsych, 2020-04-21:
@julijenv Thanks so much!
It's more than clear explanation!
By the way, the red marked connector in last picture, is it standard one? If so, would you let me know the connector name?
Some strange events happened to the PSU in icub-Seoul01, and we don't have the connector...

julijenv, 2020-04-21:
Hi @iPsych , which connector are you talking about? the power cord in between the mural socket and the power supply??? or one of the provided by IIT to make connection in between power supply to power supply or power supplies to iCub???
let me know.
if it is about the power cord you can find easily some spare parts on RS components or Farnell, I've done that when I was for a month in Japan for iCubOsaka01.

iPsych, 2020-04-21:
[image]
The DB15 cable, connected to 12V parts.

julijenv, 2020-04-21:
Hi @iPsych , I see that you have the DB15, so I guess you are missing the 2 wires connected the DB15 and the power supply 12V? Am I right?

iPsych, 2020-04-21:
[image]
Hi, @julijenv, I meant the yellow marked connector. : )

julijenv, 2020-04-21:
Hi @iPsych , coul you take a picture of the back of the 2 power supplies please? thx in advance

iPsych, 2020-04-21:
[image]
@julijenv Here it is : )

julijenv, 2020-04-21:
Hi @iPsych , could you please try to fetch me on skype please my skypei ID is icub.support

julijenv, 2020-04-22:
Hi @iPsych ,
could you please attache the final resault of the wring of the back side of the power supplies, thx in advance

iPsych, 2020-04-23:
@julijenv Sure : )
The upper one is 12V and the lower one is 40V.
[image]
[image]
[image]
[image]

julijenv, 2020-06-09:
Everything looks connected as it should.

closed 2020-06-09 by julijenv