robotology/mesh-support #937: iCubErzelli01: Left wrist not calibrating properly

ddetommaso, 2020-05-22:
## Brief description of the request/failure
The left arm doesn't calibrate properly. During the calibration the left wrist is shaking probably due to loose tendons.

 

## Accompanying material

[log_pc104_yarprobotinterface_1594.txt](https://github.com/robotology/icub-tech-support/files/4668285/log_pc104_yarprobotinterface_1594.txt)

[MOV_0841.m4v.zip](https://github.com/robotology/icub-tech-support/files/4668408/MOV_0841.m4v.zip)

maggia80, 2020-05-22:
@julijenv it seems related to a weak connection of one of the two motor cables of the wrist, or problems with AEA sensor (noisy measurements? a bit strange).

cc: @violadelbono

julijenv, 2020-05-25:
One of the wires of the AEA sensor was broken and has been changed.
The j7 of the left_arm was miscalibrated and was also recalibrated

closed 2020-05-25 by julijenv