robotology/mesh-support #939: Install new motor of thumb right hand

dariopasquali, 2020-05-27:
Ciao @julijenv,
We need to install the new motor of the right thumb of iCubGenova03.
When could you come to Erzelli?

Thank you!
Dario

julijenv, 2020-06-17:
Dear Dario,
we managed to replace the motor but ended finding the sensor of the thumb (HALL sensor - NOT the one integrated inside the motor itself) broken. We resoldered it and it makes the tricks. STILL, the 2 thumbs are not rigorously symetrical at the level of behaviour - we controlled and it doesnt look like it is related to PID.
Actions for tomorrow:
- [ ] right thumb distal to restore
- [ ] close the arm cover
- [ ] retense the right fingers abductions
- [ ] recalibrate the fingers abductions for both hand.
- [ ] middle right hand plastic tube to put back
- [ ] understand the thumbs behaviour differences

julijenv, 2020-06-18:
Hi @dariopasquali , @reafrancesco ,
The problem with the finger abduction was that the magnet on both hands was not in the position on which Vmin < Vmax ( values needed to calibrate). Therefore moving the magnet and changing the PID make now the robot working as every other robot.
Meaning Closing fingers = 60 degree. Opening fingers = 0 degree

closed 2020-06-18 by julijenv