robotology/mesh-support #941: iCubGenova04 shuts down during walking

S-Dafarra, 2020-06-04:
## Brief description of the request/failure
The robot fell twice while performing few steps with the motor led blinking.

## Detailed context
The robot is walking straight. It is shaking a bit and after a while the boards shut down.

## Accompanying material

[log_icub-head_yarprobotinterface_21216_failsDuringWalking2.txt](https://github.com/robotology/icub-tech-support/files/4731260/log_icub-head_yarprobotinterface_21216_failsDuringWalking2.txt)
[log_icub-head_yarprobotinterface_21674_failsDuringWalking.txt](https://github.com/robotology/icub-tech-support/files/4731261/log_icub-head_yarprobotinterface_21674_failsDuringWalking.txt)

violadelbono, 2020-06-05:
Hi @S-Dafarra ,

From your log files we noticed that the problem was a CRC output error from the board EB24 and consequently an error in all the following boards connected.
We opened the left arm, inspected the board and its ETH cable.
After unplugging and plugging again, it works fine without any errors when launching yarprobotinterface. Probably the problem was related to a wrong positioning of this connector.

S-Dafarra, 2020-06-05:
Great, thanks! We tried the yoga a few times and there was no error in the logger related to communication issues. We will tune a little bit more the walking to reduce excessive vibrations. I think we can close this for the time being.

closed 2020-06-05 by S-Dafarra