robotology/mesh-support #942: iCubGenova07 eyelids (face-eb22-j0) stuck in maximum position

dariopasquali, 2020-06-15:
Hi,
The eyelids (face-eb22-j0)  of iCubGenova07 are in "HW_FAULT joint position outside its hardware boundaries".
We can't move them with yarpmotorgui, either after a restart.

This morning they eyelids did not move with emotionInterface (and neither directly sending raw commands to the yarpdev).
So I moved them with yarpmotorgui, at speed 10, to half of the positions range.
They did a really fast closing and opening movement and got stuck in the maximum opening position.

Thank you,
Dario

julijenv, 2020-06-15:
Ciao Dario,
if the robot starts with the eyelids already in HW_FAULT (out of the boundaries) the only way to retrieve a correct behaviour is to move the eyelids by hand (when the motors robot are off) inside the range of motion.
Could you try that please?
thx

dariopasquali, 2020-06-17:
Thank you @julijenv I did not know I can manually move the eyelids.
I close the issue.
Thx
Dario

closed 2020-06-17 by dariopasquali