robotology/mesh-support #944: The undesirable movement of the arm and legs joints

MiladShafiee, 2020-06-16:
## Brief description of the request/failure
While the robot is in the home position, the arms and legs move and it seems that arms and legs joint connection are nos so tight.

## Detailed context
When we put the robot on home position and we try to move the arms or legs by a small pressure, it seems there is some clearance and arms move with this pressure.

## Accompanying material

traversaro, 2020-08-17:
Does anyone from @robotology/iit-dynamic-interaction-control know if this is still an issue?

S-Dafarra, 2020-08-17:
Probably this was already fixed by @julijenv. There was some mechanical backlash in the legs and in the arms before running some push recovery experiment.

MiladShafiee, 2020-08-20:
> Does anyone from @robotology/iit-dynamic-interaction-control know if this is still an issue?

Hi @traversaro. As far as I remember this issue has been solved by Julian and in the last experiments that we were shooting the video we didn't have problem.