robotology/mesh-support #945: iCubGenova02 hip problems

lrapetti, 2020-06-16:
## Brief description of the request/failure
We are experiencing some problem affecting the right/left hip motors of the iCubGenova02 robot.
In particular we are experiencing:
- suspicious motion on both right/left in hip abduction/adduction directions when the robot is in position control.
- on the right hip flexion/extension we hear and feel something obstructing the motion of the motor (by putting the motor in idle).

The issue has been discussed and verified with @julijenv and @violadelbono.

lrapetti, 2020-06-16:
CC @marco-monforte @GiuliaDAngelo

We are planning to fix it on Thursday and require quite some work, let us know if you (or any other `iCubGenova02` user you are aware of) might need the robot that day.

marco-monforte, 2020-06-16:
At the moment, no one needs it on Thursday. We will update you if something changes, but I don't think it will happen 👍

violadelbono, 2020-06-18:
Good morning ☕ 

Today @julijenv and I are still in Erzelli so won't be able to fix the robot.
We will re-book the maintenance.

julijenv, 2020-06-18:
@marco-monforte , @GiuliaDAngelo , @lrapetti , I was not able to cancel the reservation for today.
Feel free to use it today, sorry for this.

julijenv, 2020-06-18:
sorry guys, we will need to work one more time in Erzelli tomorrow and I fear I wont make it to the repair. So I prefer post pone for in 2 weeks as next week Im not there in IIT

violadelbono, 2020-06-22:
Guys,
I just booked the robot for this afternoon for the fixing.

violadelbono, 2020-06-23:
Ciao everyone,

The reason behind the obstruction and the noise of the right hip was a damaged ball bearing, and probably some of the spheres detached and blocked the pulleys movement.

We replaced the following ball bearing:

|IITCODE|Ver.|Qta |DEscription|
|--|--|--|--|
|3045||1|Rad. ball bearing 1Row. 7X14X4 - SKF CODE 618/7     ||

[image]

After launching again the robot the right hip is correctly working and it stays in position.
Now it's ready to go.

closed 2020-06-29 by julijenv