robotology/mesh-support #946: iCubGenova07 - Controlling the eyelids

dariopasquali, 2020-06-17:
Hi @valegagge 
we can't control the eyelids through the FacialExpression application.

## Description of the failure
Using the emotionInterface module from FacialExpression application is not possible to move the eyelids. For example, if I send "set all sad" everything changes except the eyelids.

The only way we can move the eyelids is through the motorgui.
Referring to issue #775 , it is not clear if the feature is developed or not.

My supposition is that the Robot Facial Expression board either misses a connection to eyelids joint or does not control it at all.

So a few questions:
* Could you please develop the eyelids control integrated with FacialExpression application?
* Which is the port to control the eyelids?
* Which is the communication protocol?

Thank you,
Dario

julijenv, 2020-06-17:
FYI
@marcoaccame for software, @pattacini for team-fix

pattacini, 2020-06-17:
Hi @dariopasquali 

The RFE only handles the LED's; to control the eyelids you need to send commands to the corresponding motor board, as you've experienced yourself through the `yarpmotorgui`.

This is because now we have better control of the motor part with respect to the past when only a very simple servo was employed.

dariopasquali, 2020-06-17:
Hi @pattacini, thank you for the answer.

I appreciate that now we can control the eyelids as any other joint. 
However, my problem is about the emotionIinterface module inside FaceExpression application.

From the [documentation ](http://www.icub.org/software_documentation/group__icub__faceExpressions.html),  the [configuration file](https://github.com/robotology/icub-main/blob/master/app/faceExpressions/conf/emotions_rfe.ini) and the `help` message of emotionInterface module is clear how the eyelids control should be part of the emotionInterface module.  However, it doesn't control them.

This also impacts other modules like the [iCubBlinkers ](https://github.com/robotology/funny-things)inside funny-things that is not working.

I think it is better to have all the face expression controls in a single interface, so we can change all the face with a single command.

So, I am asking you if you could extend the emotionInterface module to include the eyelids control.

Thank you,
Dario

pattacini, 2020-06-17:
The documentation you're referring too doesn't include the latest modifications we made a couple of days ago (see https://github.com/robotology/icub-main/pull/660), which highlights how the RPC I/F is already quite different and back-compatibility cannot be guaranteed – @Nicogene correct me if I got this wrong.

Such modifications have been pushed into the `devel` branch, but we can expose them already from `master`, if required.

With the latest HW, we can also have precise control of the eyelids that wasn't available at all before, thus this can somehow motivate the split in responsibilities. Let me check with my colleagues.

That being said, I can understand that being unable to use e.g. `iCubBlinker` with the RFE might be annoying. So, stay tuned.

pattacini, 2020-06-22:
@dariopasquali just a quick update on this to let you know that we've officially filed the request in our backlog and we'll commence working out this at the beginning of July.

dariopasquali, 2020-06-23:
That is great!
Thank you very much for your support!