robotology/mesh-support #947: Cannot control joints 8 to 15 on iCubGenova01 right arm

xEnVrE, 2020-06-23:
## Brief description of the request/failure

As per the title, it is not possible to control the joints of the right hand fingers (i.e. `r_thumb_oppose`, ..., `r_pinky`).

After running the yarpmotorgui, the situation is the following:

[image]

i.e., some of the are in hardware fault and the remaining are in idle. After pressing idle on the red ones, they do not switch to idle.

## Detailed context

I am not aware about the situation during which the issue arised, the robot was found as described in the issue.

traversaro, 2020-06-23:
Interestingly, back on friday those joints were working and were used to have the iCub grasp a small glass:
[image]

xEnVrE, 2020-06-23:
@julijenv I updated the description with the log from the `yarprobotinterface`

xEnVrE, 2020-06-23:
I restarted the robot now for other experiments and everything is working. Then, it seems that the calibration might occasionally fail. This is something that never happened in the past, at least with this robot. 

I'll keep you posted.

violadelbono, 2020-06-25:
Looking at the errors from the log files:

      2296 203,261539 <ERROR> Parameter networks use deprecated syntax 
      2335 203,524459 <ERROR> motorPwmLimit not found 
      4149 312,089017 <ERROR> getCurrentRangeRaw not yet implemented for CanBusMotionControl 

In my opinion, it sounds like some config files are old and perhaps the robot needs an update.
Keep us posted!

cc @julijenv

xEnVrE, 2020-06-26:
Well, I tried again today and I found that the calibration works properly only after the first boot of the robots boards (i.e. those controlling the joints). Then, after that, if you try to close the `yarprobotinterface` and re-open again, the calibration of the right hand fingers will fail.

@julijenv @pattacini

pattacini, 2020-06-26:
@marcoaccame do you have any clue as to why rebooting the boards may impact the behavior @xEnVrE reported just above?

pattacini, 2020-06-29:
Actually, being a CAN robot perhaps @randaz81 might have useful hints.

julijenv, 2020-06-29:
Dear All,
I think the only possible meaning is the loose CAN line in between MC4 and MAIS board. I don't think it is such a big issue, just recrimp and you will not have any more this flaky line and uncontrollable behavior.
I am on meeting today so I will be right back to you tomorrow.

pattacini, 2020-06-29:
Good if this will be of an easy solution as you said @julijenv, although from the description @xEnVrE reported it seems there exists a deterministic correlation w/ the power cycle of the boards.

julijenv, 2020-07-02:
> > 2335 203,524459  motorPwmLimit not found
> 
> Note that some of those errors were kind of expected on CAN-based robots. As having an error that is always there is not really helpful, we tried to address some of them in [robotology/icub-main#661](https://github.com/robotology/icub-main/issues/661) , however there may be more work necessary (for example I don't think we have handled the "Parameter networks use deprecated syntax", as I was not really sure about those.

Effectively it would be niced to get rid of he known error (which are indeed not error as per se) i n order to clean up the log file (red line on yarplogger) to directly spot the real problem.
@violadelbono and I are right now working on debugging the real issue.
NB: a note aside, the laptop pc were kind of stuck (yarpmanager had all the lateral and upper general commands greyed as non responsive). we reboot the 2 laptops and the robot and now it is better responsive.

julijenv, 2020-07-02:
Hi @xEnVrE , could you please confirm me if the Right last fingertips (pinkie) was or not functionning?
thx in advacne

xEnVrE, 2020-07-02:
Hi @julijenv, are you referring to the fingertip, i.e., skin taxels?

violadelbono, 2020-07-02:
The random behavior at the boot was fixed and the joints 8 to 15 are not in HW fault anymore.

The reason was a loose connection at the level of the MC4 boards in the right arm. Indeed the LEDs of some boards were off or blinking irregularly.

We checked the connectors and substituted some crimps. Now the yarplogger does not show any errors (except for the old known ones).

closed 2020-07-02 by violadelbono