robotology/mesh-support #948: Skin of right hand not working

xEnVrE, 2020-06-26:
## Brief description of the request/failure

After opening the `skinManager` and connecting the viewers, there is no response when the taxels of the right hand and/or the right hand palm of iCubeGenova01 are pressed.

violadelbono, 2020-06-26:
I checked the robot skin and the right pinkie fingertip and right palm were not working.

- For the fingertip I noticed that a cable of the sensor was disconnected from its board so I re-welded it 

-  For the palm, all the cables were well connected but I noticed that the patch is damaged were there is the grounding part so I believe that it was not in its correct position and thus not making contact (see picture below)

[image]

I tried to put it in a better position as a temporary solution.

However, after re-launching the robot, the skin of the right arm does not work anymore and this could be related to the calibration problems of the robot mentioned in the current issue [#947 ](https://github.com/robotology/icub-tech-support/issues/947)

violadelbono, 2020-07-02:
Hi @xEnVrE,

The skin of the right arm is now working. 
We checked the connections and re-welded the power cable of the palm as well. Some of the connections at the level of the boards in the arm cover were loose.

closed 2020-07-02 by violadelbono

xEnVrE, 2020-07-02:
Thank you!

cc @2103simon