robotology/mesh-support #949: both hands + left forearm skintaxels not working with old MTB

2103simon, 2020-06-26:
## Brief description of the request/failure
Taxel read out not possible. No response from the skin taxels at the fingers, thumbs and palm of both hands. Read-out with the new MTB (event-driven) works on the right hand for thumb and palm, but NOT for the fingers) .

## Detailed context
I noticed that the blue LED at the forearm not blinking, but constant shining (see attached pictures). 

## Accompanying material
[image]
[image]

julijenv, 2020-07-03:
Hi @2103simon , R U sure that you did not confuse your right with the right or the robot??
from my knowledge the event-driven skin is on the LEFT arm not the RIGHT....

2103simon, 2020-07-03:
Your right, the event-driven skin is on the left, but the sample based one is not working on non of the sides...

violadelbono, 2020-07-03:
Hi Simon,

- Right arm: the hand had some detached cables and there was also some glue blob that we removed. We also re-crimped some skin connections. Now fully working

- Left arm: it is now working without the forearm skin, we will investigate the problem on Monday. Probably some loose connections

2103simon, 2020-07-03:
Oh super! You did great work! Thanks a lot :)
The forearm is not of interest for me, it was just mentioned because I realized it too.
Thanks again!

violadelbono, 2020-07-06:
Hello @2103simon, FYI:

The skin chain (upper arm + forearm + hand) of the left arm does not work. The single components work alone, meaning that there is not enough power to make the whole skin of the arm function. This problem could be due to the fact that the skin is old and damaged.

As a solution, we deactivated the upper patch of the forearm, so now the hand, lower forearm and upper-arm skin is fully working.

2103simon, 2020-07-06:
Hi @violadelbono ,
These are good news! Not that the hole chain is not working, but that the hand works when upper arm skin is unplugged is fine for me! 
I will be at IIT on Wednesday again to check it personally. If it is ok for you, I would give a feedback when I've get an impression on my own :)

violadelbono, 2020-07-06:
Perfect! If everything will be ok feel free to close this issue

closed 2020-07-20 by julijenv