robotology/mesh-support #950: skin taxels at the fingertips not working with new MTB

2103simon, 2020-06-26:
## Brief description of the request/failure
The read-out of the skin at the fingertip of the right hand (where the new MTB is installed) ist nor working. Only the read out of the thumb and the palm is working.

## Detailed context
I saw a difference in the order of the cable at the connectors (see attached pictures). When connecting the new MTB to the sink, only thumb and palm produce output, but NOT the fingertips. When attaching the sensor output from the thumb to fingers and versa visa, the visualizer shows the output from the thumb taxels at the index finger (again, thumb works) but touching the finger no output at the thumb can be seen. The same if palm and fingers are shifted. That indicates, that the lines from sensor to MTB are working, but the taxels at the finger or the processing of the fingertips is broken.

## Accompanying material
[image]
order of the old MTB: red, green, yellow, blue, white and black
order of the new MTB> red, green, blue , yellow, white and black
blue and yellow are switched.

julijenv, 2020-07-02:
Hi @xEnVrE , could you please confirm me if the Right last fingertips (pinkie) was or not functionning?
thx in advacne

julijenv, 2020-07-03:
Hi @2103simon ,
by taking a look at the right hand connector, it looks identical to the left old MTB connector (btw the only one that we support ) - the flying one (new MTB) is hand made by you guys and I am not responsible for the pinout, so I guess that you need to figure out what pinout suit your need.
So I guess that if you plug your connector in the order that we recommand (old MTB) you find yourself with the entire fingers working.
let us knwo

2103simon, 2020-07-03:
Hi @julijenv ,
is the old mtb working already at both hands?
I changes the order of the pins already, but it changes nothing at the result. Output is only available from thumb and palm, not the fingers.

violadelbono, 2020-07-06:
> Hi @julijenv ,
> is the old mtb working already at both hands?

Yes :)

julijenv, 2020-07-06:
Hi @2103simon ,
from our side, both fingertips and palms are working just fine if you disconnect one patch of the chain skin (upper arm, forearm, and hand). We chose to disconnect the upper part of the forearm. An extensive work would need to be done in order to restore the entire responsiveness of the entire skin but unfortunately, it means rebuild patch, rebuild part of the connection, and require more than a workaround.
If you need to have the entire skin working, we will need to take the robot for a full day to restore the entire cabling and find out the responsible patch(es) that draws an extra consumption of current and weaken the closest patches and this takes a long time meaningless to say that rebuild patch take an extra step and time to be done and reglued on the plastic cover which could lead to some extra days of non usable skin part of the robot.
cc @ChiaraBartolozzi

2103simon, 2020-07-06:
Hi @julijenv ,

for me it's fine like it is now, but are you sure the skin is working as expected? Honestly it sounds not pretty good to me with "energy loss" at an unfindable spot... What ever is going on, it should be fixed, right?

As I said, for my experiments I only need the skin on the hand, the rest is not from interest.
Thank you very much for your super fast help, also @violadelbono !

Best

closed 2020-07-20 by julijenv