robotology/mesh-support #953: iCubGenova03 - right hand little fingers not moving

dariopasquali, 2020-06-30:
Hi @julijenv,
After your intervention of issue #939 we noticed a new issue with iCubGenova03 right-hand little fingers.

The pinky&ring fingers of right hand do not move.
The motor just spins empty without moving them.

I checked from `yarpmotorgui` and the encoder works fine.
So, I suppose it is a broken cable.

julijenv, 2020-06-30:
Hi @dariopasquali ,
from what I recall you had the chance to undergo a little training for repairing distal joint.
That could be the right moment for you to try if you remember something about it.
btw we have already a website about restoring disal joint that you can refresh your memories with.
here is the link:
https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#ring-and-pinkie

violadelbono, 2020-07-27:
Cables of little and ring fingers were broken and have been repaired.
The proximal thumb was also broken and has been replaced.

closed 2020-07-28 by julijenv