robotology/mesh-support #954: iCubLausanne02: yarprobotinterface segmentation fault after moving the robot

MichaelBombile, 2020-07-01:
Hi @julijenv,

After moving the robot within the lab, I noticed that yarprobotinterface is now failing to start the robot with a segmentation fault. You can see the attached log file. Several boards have timeout. I am not sure if it is due to a hardware problem. 
Could you help us resolve the problem? 

Thanks

[log_error_iCubLausanne02_01_07_2020.txt](https://github.com/robotology/icub-tech-support/files/4858216/log_error_iCubLausanne02_01_07_2020.txt)

julijenv, 2020-07-02:
Hi @MichaelBombile ,
I didnt see and i dont konw what you disconnected from one room to another but it looks like that most of the boards are not responding (MOST but not all, strangely enough).
My first guess would be to check that both the back cord and the plug to the power supplies are correctly plugged in.
Secondly, when turning ON the motors jsut wait time enough (30sec.) and double check that the overall LEDs onto the arms adn legs start booting and reach a continuous behaviour (LEDS fixed or flickering at around 1hz).
Relaunch the robot and send me the log file if it doesnt start again.
let me know

MichaelBombile, 2020-07-02:
Hi @julijenv ,

Thanks for your reply.
I proceeded as you suggested, the robot still did not start (see the log file). 
Also, after running the FirmwareUpdater and discovering the boards, it looks like some ems boards don't appear. 
I also suspect that a line could be broken.

[log_error_iCubLausanne02_02_07_2020.txt](https://github.com/robotology/icub-tech-support/files/4862976/log_error_iCubLausanne02_02_07_2020.txt)
[image]
[image]

julijenv, 2020-07-02:
Hi @MichaelBombile ,
Now it looks more clear to me.
we have evey ehternet board showing from head to right arm but not the one following the right arm (on the daisy chain, here come by order of missing : torso, left leg and finally right leg).
AS previously mentionned could you just tell me if the board beneath the cover of the tighs, (legs in general) and the bottom are enlightened.
If yes it means that it is indeed the ethernet line that is broken in between the right arm and the torso (all boards are located beneath the bottom (ass) cover).
Let me know

MichaelBombile, 2020-07-02:
Hi @julijenv,

All boards beneath bottoms are enlighten. So, this could mean that the line is broken.

julijenv, 2020-07-02:
Hi @MichaelBombile ,
So you have now 3 possible places where a broken cable can be found. Everything is located on the lineE29.
[image]
- blue circle is the luckiest one for you to check (it is a junction that you could find directly on the back of icub when the robot is bowed 90 degrees. (at the base of the bundle of cables going from upper body to bottom cover - you may have to remove the bottom cover to inspect and find it - it is labelled E29)
- orange circle is the middle easy to find out it is also located beneath at the end of the cable E29 attached to the board EMS5.
- red circle is located inside the upper right arm and you need to remove the cover of the upper arm to do that. Please follow the guide for removing the upper arm cover from :
https://icub-tech-iit.github.io/documentation/icub_tendons/remove/#22-how-do-i-operate-for-lower-arm-v2
let me know

MichaelBombile, 2020-07-02:
Hi @julijenv,

I will check and let you know.

Thanks

MichaelBombile, 2020-07-08:
Hi @julijenv,

Thanks for your instructions. Indeed, I found a broken connection on line E29 and I managed to fix it. The robot can start now. However, the robot has now a hardware fault in one of the right hand joint (see in the attached log file).
Moreover, I noticed that when the power supply is turned on, strangely the legs boards go on as well even if the main switches are still off (Switches of CPU and Motors OFF). 

[log_error_iCubLausanne02_08_07_2020.txt](https://github.com/robotology/icub-tech-support/files/4890234/log_error_iCubLausanne02_08_07_2020.txt)
[image]

julijenv, 2020-07-09:
Hi @MichaelBombile ,
unfortunately there quite a lot of error messages into your log file and it is quite hard to investigate this way.
therefore i will make the question and you will give answer and later i will be bask to the log.
First, which joint of the right hand is in hardware fault (if joint j7, please close a bit more or open more - put it in mid range and try agin to relaunch)?
I got so many AEA error due to spikes that I am quite positive that there is a problem with the ground somewhere on your robot.
And the fact that the legs remain unlightend even if the motors switch is turned OFF.
Is it possible that you pinched some cables when you close back the pants cover? I owuld suggest you to remove the cover and retry to see if the lights turn ON after having removed the cover.
let me know

MichaelBombile, 2020-07-09:
Hi @julijenv,

Thanks for your reply.
The faulty right arm joint was joint6 (wrist yaw). By putting it in the mid-joint position when the robot starts, it is now working. 
Regarding the legs lights. As soon as I turn ON the power supply, the LEDs also light up despite the fact that the motors switch is OFF. I thought first that this might be a problem with the switch (may be a short-circuit), it turns out that the switch is fine. Can this be related to the setting of the power supply? (no cabling was changed and its settings can be seen in the attached photo).

Regarding the spikes, I checked the cables after removing the cover, they seem fine. But the AEA errors still appear in the log file.

Thanks.

[log_error_iCubLausanne02_09_07_2020b.txt](https://github.com/robotology/icub-tech-support/files/4896335/log_error_iCubLausanne02_09_07_2020b.txt)
[image]

julijenv, 2020-07-09:
Hi @MichaelBombile ,

could you take a picture of the power supplies after turning OFF the motor switch? (is it already the one sent previously? the values of the power supplies looks OK if both switches are ON).

Do you remember having this issue before moving the robot?

Could you please check with a multimeter the voltage in between a black cable (get one from the hand - maybe beneath the dorso cover - disconnect the MAIS board connector) and any metallic part of the robot's frame.

voltage in between black and frame:
- If 0V (double check with diode testing on the multimeter and if it beeps), a ground (black cable) from somewhere in the robot is touching the frame.
- if the voltage is far from 0V ( >1v) one red cable is touching the frame.
Let me know

MichaelBombile, 2020-07-09:
Hi @julijenv,

Herewith the pictures of the power supplies when they are just turned on (picture 1) and after setting the top power supply to "local" and "Output On" (picture 2). Before I did not need to do such setting for the voltage to go to 39.15V (it was automatically outputting 39.15V). Moreover, it essentially when I used the above settings ("local" and "Output On") that the legs light go ON even if I haven't switched ON the motors.

Regarding the voltage between a black cable and the robot metallic part, I measured 1.2 V.

[image]
[image]

julijenv, 2020-07-09:
Hi @MichaelBombile ,
Please leave the top power supply on "Remote " (the third button on the power supply panel. I didn't pay attention to this before but the problem is that you put "local" meaning that you don't control by the motor switch the output of the power supply.)
let me know

MichaelBombile, 2020-07-09:
Hi @julijenv ,

When I put on "Remote", the voltage drops since the power supply goes automatically to "Output Off" and I does not got to "Output On" even if pressed. It only goes to "Output On" when is in "Local" mode

julijenv, 2020-07-10:
hi @MichaelBombile ,
it is normal that it drops to 0V as it is operated from the motor switch (the block switches - CPU AND MOTORS). Turning On the switch motors (not the one from the power supply, this one should always be ON) as the power supply 40V is in "Remote" it will start pouring voltage adn current into the boards controlling the robot motors.

MichaelBombile, 2020-07-10:
Hi @julijenv,

Thanks for your reply. It was my mistake. I realized that there was no problem with the power supply yesterday evening. 
The problem that still persists is the AEA encoder that has spikes (from BOARD 10.0.1.6). 
Do you have any suggestion?

Thanks

julijenv, 2020-07-10:
Hi @MichaelBombile ,
could you please send me the last yarprobotinterface please thx in advacne

MichaelBombile, 2020-07-10:
Hi @julijenv,

Herewith the log file
[log_error_iCubLausanne02_10_07_2020.txt](https://github.com/robotology/icub-tech-support/files/4902280/log_error_iCubLausanne02_10_07_2020.txt)

violadelbono, 2020-07-10:
Hello @MichaelBombile,

Looking at the last log file you sent:

- There are some errors in the Left Arm calibration: it seems that the calibration of joint 10 (thumb distal) failed, so we suggest to check if the cable is broken or it was somehow wrongly positioned at the starting phase. You can follow our repairing [manual](https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V2/#thumb). In particular, it shows how to manually rotate the motor in order to check if the finger closes.

- Right leg: the AEA spikes can be due to a shield cable damaged or detached, thus causing additional noise. This is not a huge problem, but you could check the sensor cables by opening the cover. However, the main problem in the right leg is this error message

      [ERROR] from BOARD 10.0.1.8 (right_leg-eb8-j0_3), src LOCAL, adr 0, time 56s 37m 102u: (code 0x04000003, par16 0x0000 par64 0x0000000000000000) -> DEBUG: tag03 + strain timeout

Could you please check if FirmwareUpdater detects the board 8? If yes, you should compare its firmware version with board 7 (located in the other leg)  and see if there are differences.

Let us know

MichaelBombile, 2020-08-04:
Hi @violadelbono,

Thanks for your suggestions and sorry for my late reply, I was on vacation.

Regarding the errors in the arms calibration, we have indeed some broken tendons in the hands. We are planning to fix them later and if we have any problem, we will open another issue.

The error message from board 10.0.1.8 has been addressed and it seems fine now. The AEA spikes still persist but I suppose it is not a big problem.  The new log file and the FirmwareUpdater  screenshot are attached.

Could you please let us know if anything else (apart from the fingers) should be immediately addressed. Otherwise this issue can be closed.

Thanks. 

[log_iCubLausanne02_2020_08_03.txt](https://github.com/robotology/icub-tech-support/files/5021067/log_iCubLausanne02_2020_08_03.txt)
[image]

closed 2020-08-09 by MichaelBombile