robotology/mesh-support #956: Lyon/Dijon iCub shoulder cable broken

pfdominey, 2020-07-09:
## Brief description of the request/failure
Doing some full upper body movements - happy to see iCub at work!   After a collsion between the right arm and a table, the robot went limp, and inspection revealed a broken cable in the right shoulder.  I believe it is cable 7.  I include some photos and I'd like your confirmation that it is indeed cable 7.  We have some cable 7s here.  Can you take a look at the photos and let me know if it is cable 7, and remind me of any details on how to do the replacement.  Thanks!

## Detailed context
 I was running ctpservices in seperate terminal for leftarm, rightarm, head and torso, and then control-c'ed the left and right arm ctpservices, and verified that the arms no longer were moved when a script called them.  But then the arms started moving again and hit the table.
[image]
[image]
[image]
[image]

## Accompanying material

julijenv, 2020-07-09:
Dear Pf. @pfdominey ,
nice to hear that the robot is in use and work even if it got a hard time colliding with table.
You are perfectly right, it looks like the 7 is out of place, but it isnt broken even if quite nicely pinched.
Either you decide to replace it, or just remount it and see if by chance the cable stands (as it is not broken), you ill now have the chance to follow our brand new on-line documentation for repairing iCub.
please go to the following [link](https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/#important-notes-for-cable-from-8-to-5_1).
let me know

pfdominey, 2020-07-09:
Hi Julien,
Nice to hear from you! Thanks for your quick response.  I hope things are going OK in Genoa.
Yes, it seems that the cable is not broken.  but well worn.
The on-line documentation is great!  
So it should be relatively easy to remount it?  A few questions:
1. the cable is already in place, so I dont need to loosen things to make space for the crimps, right?
2. so I should reposition the two crimps "jumped over the two teeth holding them in position"
3. then the docuementation says to tighten the  screws of the tensioners 7 and 8 - where are these screws?  and should I loosen them first?

Thanks
cheers
Peter

julijenv, 2020-07-09:
Dear Pf. @pfdominey ,
1. from the picture it looks like the cable 7 did jump from its route - so unfortunately you will need to loosen the external part to first reposition the cable on the right route.
2. yes, postion the 2 crimps ont their respective tooth and make sure the 2 cables are routed correctly (along the pulleys).
3. yes, you first need to untighten the 2 tensing screws holding the one side of the cable crimp (of course the one not hang on to the tooth)
[image]
[image]
Let us know,
best of luck for your first shoulder rewiring.
cheers,

pfdominey, 2020-07-09:
Great!
Today was my last day in the lab for a while, so I will do the repair several weeks from now.  If it is better for you to close this issue now, that is ok with me, as I think you have fully responded.  Then, when I do the repair if I have additional questions I can open a new issue.  Or if you want to leave this one open until I do the repair that is fine as well. Thanks again. cheers - Peter

pfdominey, 2020-08-31:
[image]

pfdominey, 2020-08-31:
Greetings, I am back fixing the right shoulder.  I removed the pulley and loosed the two screws so as to have room to put cble 7 back in place.  Now cable 8 is loose but can be put back in its seating indicated by the blue arrow  The green arrow shows cable 7 wich winds up and behind and will come out on the right.  It should insert in the pulley marked with the red arrow, right?  It seems that I need to rotate that pulley in a clockwise direction (in this figure) so that the cable head can reach into its slot.  How can I rotate the pulley for cable7?
Thanks

pfdominey, 2020-08-31:
Greetings again,
I got the cables installed, then, the hex nut for the second tensing came loose and fell off and I have lost it...
The cables are thus loose, and cable 7 slipped off its pulley, so I have to start over.
I am not sure to be able to do this by myself.
Do you know if there is any way we could get someone here to help me ?  Maybe Serena in France ?  I am in Dijon.
Please let me know how we can proceed.
cheers
Peter

pfdominey, 2020-08-31:
ps - For anyone who would be able to come to Dijon to fix this, we would of course pay for any travel expenses.

julijenv, 2020-08-31:
Dear Professor,
I think if you have a bit of time to spare on skype, I guess that i should be able to explain you how to restore the wiring, without having anyone but yourself to restore it.(under my guidance obviously).
let me know
my skyoeID is icub.support

pfdominey, 2020-08-31:
Hi Julien,
ok we can try!!  Buy I did loose the screw for the tightener  " the hex nut for the second tensing came loose and fell off and I have lost it..."  
Is there another tightener in the foot or somewhere that I could use, if I cant find that screw?

Do you have time tomorrow to try to help me by video??

thanks

cheers

Peter

julijenv, 2020-09-01:
Dear Pf. @pfdominey,
the screw M2.5x6 you lost is a hexagon socket cap head, 
[image]
I'm pretty sure if you sweep the ground to a radius of 1.5m around the robot you should find it (or else the screw just fell down inside the upper arm cover, sometimes, inside the butt cover), a little check-in into those areas should make you find it.

Unfortunately, the screw you need is just used into the tensioners inside the arms so alternatively, just pass by any local hardware store - leroy Merlin, castorama - or just RS or amazon, should have you the piece in a breeze of time.
At that point, we may proceed together and rewire with the help of skype.
Please reach me on skype and I'll be helping you.
best regards professor,

pfdominey, 2020-09-01:
Hi Julien,
I found the screw!  Can we try to meet by skype tomorrow?  Would you be available around 9:30?
I tried sending you a message by skype.
cheers
Peter

pfdominey, 2020-09-02:
[test.txt](https://github.com/robotology/icub-tech-support/files/5161697/test.txt)
After a heroic effort by Julien, we were able to replace cable 7.
The yarprobotinterface would not run.
Attached is the ouput file

pfdominey, 2020-09-02:
I tried a total shutdown and restart.  yarprobotinterface does not run correctly and the robot does not do its inital move to a neutral posture.  As I mentioned.  IN july after breaking the right arm, I started the robot again and it started up correctly, without the right arm.  So I dont understand why it doesnt start now.  I have some meetings now, and will get back in touch with Julien later this afternoon.  Thanks

pfdominey, 2020-09-02:
Hi Julien, 
It is 17h, maybe you are gone.  I tried to start the robot again, and it does not work.  I have got team viewer connected .

pfdominey, 2020-09-02:
[test2.txt](https://github.com/robotology/icub-tech-support/files/5163252/test2.txt)
here is the log from running again yarprobotinterface

pfdominey, 2020-09-02:
Hi Julien - is this the crimper we need? (see photo)
[image]

julijenv, 2020-09-03:
Dear Pf. ,
No unfortunately it is not the crimping tool (or at least not one of the dedicated one I know or used) I think it is surely a coating remover.

violadelbono, 2020-09-08:
Hello Prof. @pfdominey 

In order to send you the components you need, could you please write me you email address?

pfdominey, 2020-09-30:
Hi Julien,
I have recieved the CAN filter - shall we connect by skype to make the repair?
Thanks
cheers
Peter

pfdominey, 2020-09-30:
Hello - anyone there?  I would like to have some verification of my connections before proceeding.  Let me know if anyone can here me!

pfdominey, 2020-09-30:
I close this because the cable is replaced - thanks !!

closed 2020-09-30 by pfdominey