robotology/mesh-support #957: iCubGenova02: right shoulder cable broken

lunagava, 2020-07-10:
## Description of the request/failure

Ciao,
I'm sorry but I just broke the purple iCub right shoulder cable moving the joint close to one of its limits during some experiments.

Should be the "shoulder rotation cable N1".

## Accompanying material

[image]

violadelbono, 2020-07-13:
Hi @lunagava 

There were actually two broken tendons:  # 2 and # 7. 
They were taken from the warehouse and replaced:

| Item / IITCODE | Quantity |  Description | Notes  |
|:---:|:---:|:---|:---|
| 4759 | 1 | 	SHOULDER ROTATION CABLE 2 |  |
| 4764 | 1  | SHOULDER ROTATION CABLE 7 |  |

We also noticed some friction in the block of pulleys in the shoulder and added grease.
The robot is now functional and ready to be used.

lunagava, 2020-07-13:
Hi @violadelbono. 
Thank you for your early intervation. I am glad that iCub is ready to be used.

closed 2020-07-13 by violadelbono