robotology/mesh-support #958: iCubGenova07 - Right Hand little fingers calibration fail - Left Eye cable issue

dariopasquali, 2020-07-10:
Ciao @julijenv and @violadelbono ,
We have a few other issues on iCubGenova07

## Right hand little fingers calibration fail

During the calibration the robot closes the **pinky finger but not the ring finger**.
Even if those fingers:
* Are supposed to be paired
* Should not move during the calibration

After the failure, the joints of right hand are in various states:
* Joint 10 (thumb distal), Joint 12 (index distal) and Joint 14 (middle distal) are **Idle**
* Joint 15 (little fingers) is **Not Configured**
* The others are running

On `yarprobotinterface` i just receive the error for `Right_Arm_Calibration : Timeout while calibration 15` but no `HW_FAULT`

I don't know if this is clear, I attach a GIF of the whole process
[image]

## Left Eye Camera not working

The Left eye camera cable is probably broken or not connected.
The camera does not work at startup, but tapping the head it ""fixes"". However, this hack works until the next startup.

dariopasquali, 2020-07-21:
Up

violadelbono, 2020-07-28:
Hi @dariopasquali and @reafrancesco,

Here is a recap:
- Cables of little & ring fingers were replaced and now they are paired.
- Thumb oppose is broken (motor spins for a while without closing the finger) and still need to be fixed -we will do it this Thursday
- Cameras: the flat cable of the left eye was replaced. Regarding the right eye, the camera works but it looks a little blurred. It is possible that also its flat cable is damaged, so we could replace it as well on Thursday.

violadelbono, 2020-07-31:
30/07: the thumb oppose joint was repaired. 

The following cable was taken from the Toolbox:

| IIT CODE  | ALIAS  | QUANTITY  | DESCRIPTION  | 
|---|---|---|---|
| 1815  |  CG077063 | 1 m  |Stainless steel microcable, cable construction 7x7mm, nom. dia 0,63mm, not coated, AISI 316|

dariopasquali, 2020-08-04:
Hi @julijenv and @violadelbono ,
after your intervention last week we noticed several issues:

- [ ] Right thumb distal closes a few seconds after the startup. I suppose the cable is rolled up in the opposite direction (the encoder is ok, the thumb closes if I move it manually)

- [ ]  Right thumb oppose does not move even from yarpmotorgui

- [ ] If I put idle the thumb oppose, also the thumb proximal goes idle

- [ ] Right little fingers don't move to their maximum or don't move at all. I wasn't able to properly reproduce this issue.

- [ ] Right arm calibration is odd, the two arms are not symmetric. However, If I open the iCub-Gui, they appear as symmetric

- [ ] During the shutdown iCub fells down on the back instead of gently lean on. On the logger, we have an error on the rfeimu `failure on reading the data from sensor rfeimu_gyro, no data will be sent on the port`.  

We also have an update on issue #967 

If you have time, could you fix it tomorrow?

julijenv, 2020-08-31:
we updated the ticks above, coud you please tell us about the missing ones, because from what we recall we didnt notice that matter on yuor boto after having made the repair.
let us know.

> * [ ]  Right little fingers don't move to their maximum or don't move at all. I wasn't able to properly reproduce this issue

I think this one has been reset for mimicking or being symetrical.

> * [ ]  Right arm calibration is odd, the two arms are not symmetric. However, If I open the iCub-Gui, they appear as symmetric

We also notice that the right arm was off compared to the normal icub-end-of-calibration-phase positon.
WE double check with yarpmotorgui, and it appears that the value was different from the normal one but you suspect that you made this change for an experiment - or a specific reason.

> * [ ]  During the shutdown iCub fells down on the back instead of gently lean on. On the logger, we have an error on the rfeimu `failure on reading the data from sensor rfeimu_gyro, no data will be sent on the port`.

Unfortunately, as therewas the thumb oppose broken, it is hard to tell if the complain is really related to rfeimu or something else, when the robot shutdown (ctrl-C or stop) evrything can be expected to happen when the robot is not perfectly working.
We will check this next time.
In the meanwhile as probably the new FW release would be done when we will pass by, I advise you to dowload and install the new FW for the rfeimu, i know that it has been made quite a few debug on it lately.

dariopasquali, 2020-09-12:
Hi!
I ran a few more tests on the robot yesterday.

## Right Arm Calibration
> * [ ] Right arm calibration is odd, the two arms are not symmetric. However, If I open the iCub-Gui, they appear as symmetric

The calibration is still asymmetrical (see picture).
However, the software assumes it is perfectly normal. For example, moving the Joint 3 to its maximum pushes the joint over its limits raising an HW_FAULT due to i2t overcurrent. The same for the other joints, they are misplaced but for the yarpmotorgui they are symmetrical. For what I recall we did not change the calibration settings, just the software limits. 

[image]

## Right Thumb
The right thumb is not working properly however it's not clear to me how to replicate the issue.
Here is the story:
After the aforementioned Joint 3 HW_FAULT, I put the joint in idle in order to move it.
While I was touching the to skin sections of the arm suddenly the **right thumb distal** closed to its maximum.
I put it idle with the yarpmotorgui and opened it.
As I run it again it wen to its maximum.

Right now, **with the motors turned off, moving it manually** everything is super tensed (see attached video here [right thumb.zip](https://github.com/robotology/icub-tech-support/files/5212951/right.thumb.zip)):
* A slight movement of opposite joint moves also the distal
* I cannot close the proximal joint at all
* If I open the proximal joint, the distal is slightly closed and tensed

## Wrist Wiring
One of the cables that control the fingers is not in the same band with the other cables. I attach two pictures because maybe it's easier. Now I put it again where it should be, but maybe it is too long or it is related to the previous issue.

[image]

## Other issues

I did not notice these issues.

> * [x]   Right little fingers don't move to their maximum or don't move at all. I wasn't able to properly reproduce this issue

> * [x]   During the shutdown iCub fells down on the back instead of gently lean on. On the logger, we have an error on the rfeimu `failure on reading the data from sensor rfeimu_gyro, no data will be sent on the port`.

julijenv, 2020-10-15:
Hi @dariopasquali ,

  - [x] Right arm calibration is odd, the two arms are not symmetric. However, If I open the iCub-Gui, they appear as symmetric
The calibration is still asymmetrical (see picture).

we also recalibrate the right arm (the magnet was not glued as it should be - now it is), we didnt have time to copy the iCubGEnova0/calibrator/left_arm_calib.xml to your repository on github, could you please make a diff in between the one .local/etccc that we changed to recalibrate the elbo and the one $ROBOT_CODE/robotts-configuration/iCubGenova07 and just make sure that just the left_arm_claib.xml has been chenaged and later make a PR to have this change on github.
thx in advacne

reafrancesco, 2020-11-09:
Thanks after last @julijenv  @violadelbono `s intervention the issue on the right arm calibration is fixed. I proceed by closing the issue. Thanks.

closed 2020-11-09 by reafrancesco