robotology/mesh-support #959: Replacement of a plastic lever in the blinking system

carlomazzola, 2020-07-14:
## Brief description of the request/failure
iCubGenova03: break of the plastic lever connected to the servomotor mechanism of blinkers

## Detailed context
In the front cover of iCub's head the plastic lever that connects blinkers to their motor is broken. We need a spare part of it for the replacement. It broke while blinking because the cover had been attached so that it touched some points of the blinking system. 

## Accompanying material

[image]
[image]

julijenv, 2020-07-14:
Hi @carlomazzola ,
I have the piece you reuested, I will mount it next time i'll come to you guys.
Just please remind me about it.
I will let you know some days before when we shulld be passing by.

carlomazzola, 2020-07-15:
Hi @julijenv ,
Perfect! Thank you very much!
I'll remind you about that
Bye!

violadelbono, 2020-07-28:
Hello @carlomazzola,

The piece was substituted.
Let us know if it will need some adjustments.

julijenv, 2020-08-31:
still waitng to get a feedback from you guys.
@carlomazzola

carlomazzola, 2020-08-31:
Hello, sorry if I have been late.
Actually, I couldn't try the eyelids system because we think the cover of
the face needs to be a bit adjusted: it is old and it seems that the
plastic has been altered a bit over time. If I would try the system now, it
is highly probable that it will break again.
As soon as we try it, I will let you know.
Sorry again and thank you!
Carlo

julijenv, 2020-10-12:
Hi @carlomazzola ,
working or not?
let us nkow
if yes please close the issue.
thx in advance

carlomazzola, 2020-10-13:
Hi Julien,
I will check as soon as I will come back to Erzelli, and I will let you
know,
sorry,
Carlo

carlomazzola, 2020-11-02:
Hi @julijenv , 
I checked and everything is ok, thank you!
Carlo

closed 2020-11-02 by carlomazzola